cocotb/cocotb #2354: No COCOTB_REDUCED_LOG_FMT

raysalemi, 2021-01-18:
`% make help` says there is a `COCOTB_REDUCED_LOG_FMT` make variable that shortens the log format lines, but if one greps the makefiles directory there is no such string. Defining this variable makes no difference.

cmarqu, 2021-01-18:
It's handled by the Python code:
https://github.com/cocotb/cocotb/blob/aa956f83e119e4ca4615c5b8660eb15c4dfa8623/cocotb/log.py#L43

imphil, 2021-01-18:
It's not a Make variable, it's an environment variable (read by the cocotb Python code). So you'll need to set it before calling make, e.g. `COCOTB_REDUCED_LOG_FMT=1 make` (https://docs.cocotb.org/en/stable/building.html?highlight=COCOTB_REDUCED_LOG_FMT#envvar-COCOTB_REDUCED_LOG_FMT)

cmarqu, 2021-01-18:
I'll improve the docs.

closed 2021-01-20 by cmarqu

raysalemi, 2021-01-21:
Is TOPLEVEL really an environment variable? I set it in Make.

raysalemi, 2021-01-21:
Is TOPLEVEL really an environment variable?

```

+ | The following variables are environment variables:

64 | 66 |   | Cocotb
65 | 67 |   | ------
66 | 68 |   | TOPLEVEL                  Instance in the hierarchy to use as the DUT
```