cocotb/cocotb #2359: Logging level should be logging.NOTSET

raysalemi, 2021-01-19:
```

332 | # instead of `logger.debug` outside of this function becomes an
-- | --
333 | # error.
334 |  
335 | if verbose:
336 | logging.basicConfig(format="%(levelname)s %(message)s", level=logging.DEBUG)

```
Calling `basicConfig` is unfriendly to other Python code that wants to use the logging as it cannot be called twice.  If it must be called I think it should be called with logging.NOTSET

I bring this up because I have a custom log level set to 5 and I cannot see it.

eric-wieser, 2021-01-19:
You're referring to these lines, right?

https://github.com/cocotb/cocotb/blob/ccf4ba4ffb6ed6fae23e79af56df8acb001cffb8/cocotb/_vendor/find_libpython/__init__.py#L335-L336

cmarqu, 2021-01-19:
Do the last two sections in https://docs.cocotb.org/en/latest/library_reference.html#logging-reference-section help?

raysalemi, 2021-01-19:
Yup.

raysalemi, 2021-01-19:
Aha.  That could work.
Or changing the environment variable.
Thanks.

raysalemi, 2021-01-19:
Hmm.  Have tried both basiconfig force and environment variable. Am still
not seeing my 5 level messages.

raysalemi, 2021-01-19:
Also getting this error:

    cocotb.log.default_config()
AttributeError: 'SimBaseLog' object has no attribute 'default_config'

eric-wieser, 2021-01-19:
`cocotb.log.default_config` does indeed give an AttributeError, you need to use:
```python
from cocotb.log import default_config
default_config()
```

Note that `cocotb.log` _outside_ an import statement refers to the global logger instance, while `cocotb.log` _inside_ an import statement refers to the module of that name.

eric-wieser, 2021-01-19:
But note that `default_config()` is already called by cocotb at startup, so I see no reason why you would want to call it.

raysalemi, 2021-01-19:
Are there tests that show that custom levels make it through cocotb?  I
don't understand why I'm not seeing my 5-level messages.

eric-wieser, 2021-01-19:
Can you do the following?

* Produce a short python file that emits 5-level log messages. Don't use cocotb at all, just a regular python script.
* Add `from cocotb.log import default_config; default_config()` to the first line, and check if it still works. Again, run the script normally, not through the simulator
* Post the python file here.

raysalemi, 2021-01-19:
This program throws a RuntimeError "no Simulator available!"

Without `default_config()` there is no error and the "Test msg" goes to the
screen

import logging
from cocotb.log import default_config

default_config()
logging.basicConfig(level=logging.NOTSET)
logger = logging.getLogger("name")
logging.addLevelName(5, "TEST")
logger.setLevel(5)
logger.log(5, "Test msg")

eric-wieser, 2021-01-19:
@ktbarrett, I thought we fixed things so that "no simulator available" didn't happen for simple cocotb imports. Is that not released yet?

ktbarrett, 2021-01-19:
@eric-wieser Calling `default_config` calls into the simulator. https://github.com/cocotb/cocotb/blob/26304cfc5fc9adbf567f2929910d5b6b9f414331/cocotb/log.py#L109-L110

eric-wieser, 2021-01-19:
Hmm, should we add a `try` / `except` around that?

eric-wieser, 2021-01-19:
@raysalemi, does this work (not crash)?
```python
import logging
from cocotb.log import default_config

try:
    default_config()
except RuntimeError:
    pass  # no simulator, but the rest of the config should be done
logging.basicConfig(level=logging.NOTSET)
logger = logging.getLogger("name")
logging.addLevelName(5, "TEST")
logger.setLevel(5)
logger.log(5, "Test msg")
```

ktbarrett, 2021-01-19:
Or we could move it into `_initialize_testbench`? Depending on if you expect people to be calling it with any regularity without being in a simulation. It's probably fine as is and the hack you suggested will work for this case.

raysalemi, 2021-01-19:
fyi:
(base) (venv) raysalemi@WriteNow cocotb % pip install --upgrade cocotb
Requirement already satisfied: cocotb in
/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages (1.4.0)

raysalemi, 2021-01-19:
Good idea!

raysalemi, 2021-01-19:
This does not solve the problem because the exception is thrown in the
"logger.log()" call.

garmin-mjames, 2021-01-19:
The problem is `utils.get_sim_time()` which calls `simulator.get_sim_time()`, which raises the `RuntimeError`.

closed 2021-03-14 by github-actions[bot]