cocotb/cocotb #2362: cocotb colorizer does not support custom logging levels

raysalemi, 2021-01-20:
The attached file demonstrates that cocotb does not support custom logging levels. This may be the intention as there is an assertion in the code regarding acceptable logging levels, it is limiting. 

I don't think cocotb should throw an exception on an unexpected level. Instead I think it should have a default behavior. It appears the log colorization system doesn't know what to do with `5`. It likely needs a `try`/`except`.

```
Traceback (most recent call last):
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/logging/__init__.py", line 1081, in emit
    msg = self.format(record)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/logging/__init__.py", line 925, in format
    return fmt.format(record)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/log.py", line 251, in format
    msg = '\n'.join([SimColourLogFormatter.loglevel2colour[record.levelno] % line for line in msg.split('\n')])
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/log.py", line 251, in <listcomp>
    msg = '\n'.join([SimColourLogFormatter.loglevel2colour[record.levelno] % line for line in msg.split('\n')])
KeyError: 5
Call stack:
```
[bug_example.zip](https://github.com/cocotb/cocotb/files/5842227/bug_example.zip)

eric-wieser, 2021-01-20:
Does this mean you were able to solve #2359?

Either way, this is an obvious bug with a simple fix, thanks!

raysalemi, 2021-01-20:
I'm not sure yet. I'm still not seeing my 5 level messages, but I'm
thinking they must not be reaching cocotb's logger at all. I have
propagation set to False in mine.

raysalemi, 2021-01-21:
> Does this mean you were able to solve #2359?
> 
> Either way, this is an obvious bug with a simple fix, thanks - changing `[record.levelno]` to `.get(record.levelno, "")` would fix it:
> 
> https://github.com/cocotb/cocotb/blob/26304cfc5fc9adbf567f2929910d5b6b9f414331/cocotb/log.py#L250-L253

That's a good idea.  I've just learned about `get()`

eric-wieser, 2021-01-21:
Would you be willing to make a PR with that change? It would be a good exercise in getting familiar with how PRs work on github, but if you don't want to I'm sure someone else will fix it soon.

raysalemi, 2021-01-21:
I was thinking that myself.  I'm trying to learn about how GitHub works.

cmarqu, 2021-01-21:
We do have instructions (with links) about contributing:, starting ~ here: https://github.com/cocotb/cocotb/blob/master/CONTRIBUTING.md#setting-up-a-development-environment

For the small change above, starting here may be enough: https://github.com/cocotb/cocotb/blob/master/CONTRIBUTING.md#how-to-get-changes-merged

raysalemi, 2021-01-31:
I'd like to fix this. I'm not sure how to assign it to me.

I've tried setting up my environment but tox says it cannot find cocotb:

```
SIM=icarus TOPLEVEL_LANG=verilog tox -e py38-macos
Modelsim/Questa executable (vdbg) executable not found. No FLI interface will be available.
py38-macos develop-inst-noop: /Users/raysalemi/repos/cocotb
py38-macos installed: attrs==20.3.0,cocotb==1.5.0.dev250+gb49c16dd,coverage==5.4,iniconfig==1.1.1,packaging==20.9,pluggy==0.13.1,py==1.10.0,pyparsing==2.4.7,pytest==6.2.2,pytest-cov==2.11.1,toml==0.10.2,xunitparser==1.3.3
py38-macos run-test-pre: PYTHONHASHSEED='2961615516'
py38-macos run-test: commands[0] | pytest
============================================================================== test session starts ===============================================================================
platform darwin -- Python 3.8.3, pytest-6.2.2, py-1.10.0, pluggy-0.13.1 -- /Users/raysalemi/repos/cocotb/.tox/py38-macos/bin/python
cachedir: .tox/py38-macos/.pytest_cache
rootdir: /Users/raysalemi/repos/cocotb, configfile: setup.cfg, testpaths: tests/pytest, cocotb/utils.py, cocotb/binary.py, cocotb/_sim_versions.py
plugins: cov-2.11.1
collected 14 items / 4 errors / 10 selected

===================================================================================== ERRORS =====================================================================================
_______________________________________________________________ ERROR collecting tests/pytest/test_binary_value.py _______________________________________________________________
tests/pytest/test_binary_value.py:6: in <module>
    from cocotb.binary import BinaryValue, BinaryRepresentation
E   ModuleNotFoundError: No module named 'cocotb'
_______________________________________________________________ ERROR collecting tests/pytest/test_binary_value.py _______________________________________________________________
ImportError while importing test module '/Users/raysalemi/repos/cocotb/tests/pytest/test_binary_value.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
```

However it is right there:

```
(cocotb_dev) (base) raysalemi@WriteNow cocotb % pip list
Package    Version
---------- ----------------------
cocotb     1.5.0.dev250+gb49c16dd
pip        21.0.1
setuptools 41.2.0
```

So I'm at a loss.

ktbarrett, 2021-01-31:
Your print out implies that the tox env already existed, but perhaps it as corrupted? Occasionally I have to wipe the `.tox/` directory to force a refresh or I get errors. Also, tox builds it's own virtualenv with the project under test (cocotb) and listed dependencies. It doesn't use the cocotb installation in your current environment.

raysalemi, 2021-01-31:
Perhaps the problem is that I already made a venv and was running in it?

raysalemi, 2021-01-31:
Removing .tox did the trick.

raysalemi, 2021-01-31:
Does this issue close when the pull is accepted or do I close it?

cmarqu, 2021-01-31:
Since you have the "Fixes ###" line in the PR, this issue will close with the PR being merged.

PR #2382 (2021-02-02): Fixes the KeyError when the colorizer is presented with a novel log level. (In the test case 5)
If a user defines a log level the colorizer used to raise a KeyError
Now it simply doesn't provide a color as it does with DEBUG level.
I accessed the dictionary using get()

Fixes #2362
closed 2021-02-02 by ktbarrett