cocotb/cocotb #2364: Recommend checking for TOPLEVEL

raysalemi, 2021-01-21:
If TOPLEVEL is not set `iverilog` fails with an obscure error message because there is nothing after the `-s` option.  I recommend that `Makefile.sim` check for TOPLEVEL and (and other required variables) and give a clean error message.

ktbarrett, 2021-03-09:
This issue is specific to Icarus. Other simulators allow `TOPLEVEL` to be empty, which in cocotb will return the root handle as the dut object, and not some entity.

haroonshafique-10xe, 2021-05-07:
> This issue is specific to Icarus. Other simulators allow `TOPLEVEL` to be empty, which in cocotb will return the root handle as the dut object, and not some entity.

@ktbarrett I tried running it without setting `TOPLEVEL` for `icarus` as well as `verilator`, it failed for both.

ktbarrett, 2021-05-27:
Thanks to @haroonshafique-10xe this was closed by #2547.

closed 2021-05-27 by ktbarrett