cocotb/cocotb #2365: Exceptions in a Coroutine that is forked multiple times can segfault Riviera Pro

jwprice100, 2021-01-21:
CoCoTb Version: 1.4
Operating System: CentOs 7 64 [bit](url
[example.zip](https://github.com/cocotb/cocotb/files/5850119/example.zip)
)
Simulator: Riviera Pro 2019.04 64 bit
Python Version: 3.7.2 (self installed)
[simlog.txt](https://github.com/cocotb/cocotb/files/5850136/simlog.txt)

garmin-mjames, 2021-01-21:
@ktbarrett is this the same as #1347?

ktbarrett, 2021-01-22:
Duplicated issue in GHDL. [sim.log](https://github.com/cocotb/cocotb/files/5856756/sim.log). I don't think it's #1347, just another issuer with a re-entrant scheduler (which is some other issue IIRC).

First call to `cocotb.fork` fails and kills the test, but the test continues calling `cocotb.fork` again, failing again, killing the test again, where is where the error comes from.