cocotb/cocotb #2366: Why does cocotb.test() have parentheses?

raysalemi, 2021-01-22:
My linter complains that 

```
@cocotb.test()
async def hello_world():
    print("Hello, world")
```

Is missing a parameter named `f` in the decorator call. I rarely see parenthesis in a decorator. Why are there parenthesis here?

eric-wieser, 2021-01-22:
Which linter are you using? I'm not surprised that we're confusing it, but if we know the linter we might be able to teach it otherwise.

> I rarely see parenthesis in a decorator. Why are there parenthesis here?

So that you can do `@cocotb.test(expect_fail=True)`

raysalemi, 2021-01-22:
It's the one in PyCharm.

How does the test determine whether fail happened or not?

raysalemi, 2021-01-22:
A better way to ask is, "How do I tell cocotb that a test failed?"

eric-wieser, 2021-01-22:
Using `assert False` makes a test fail, if that's your question

raysalemi, 2021-01-22:
So cocotb expects AsssertionError and marks a test as failed if it sees it?

eric-wieser, 2021-01-22:
Yes, exactly

closed 2021-01-22 by raysalemi