cocotb/cocotb #2367: What is the `f` Argument in cocotb.test()?

raysalemi, 2021-01-22:
What does the `f` argument do below? I cannot find it in the code.
```
    def __init__(self, f, timeout_time=None, timeout_unit=None,
                 expect_fail=False, expect_error=False,
                 skip=False, stage=None):
```

eric-wieser, 2021-01-22:
It's the test itself, and it's just passed into the coroutine constructor:

https://github.com/cocotb/cocotb/blob/f65e08a652f5f34357bda733ca397f657a73a3c0/cocotb/decorators.py#L572

There's some magic in place to ensure that `isinstance(some_test, cocotb.test)` true, which is documented here:

https://github.com/cocotb/cocotb/blob/f65e08a652f5f34357bda733ca397f657a73a3c0/cocotb/decorators.py#L429-L449

raysalemi, 2021-01-22:
Oh, right. Regular decorator behavior.

Personally, I would have let the test-writer decide on the definition of
fail and leave the decorator as simple as possible, but of course that's
cheap engineer talk.

closed 2021-01-22 by raysalemi