cocotb/cocotb #2370: Handling `z`

raysalemi, 2021-01-24:
Sometimes my signal is set to `z` at the beginning of a simulation, so I do this:

```
            await FallingEdge(self.dut.clk)
            try:
                done = int(self.dut.done)
            except ValueError:
                done = 0
```

I have to do this because sometimes `dut.done` has been set to an integer (which has no`value`) and sometimes it is `z`. 

Is there some better way to safely convert a signal to an `int`?

eric-wieser, 2021-01-25:
> I have to do this because sometimes dut.done has been set to an integer (which has novalue)

I don't understand what you mean by this. Do you mean that `type(self.dut.done)` gives `int`? If that's actually happening, then either your code is doing something wacky, or there is a bug in cocotb.

raysalemi, 2021-01-25:
I set to 0 as an initialization and this is an int.

Should it be getting converted to something else?

eric-wieser, 2021-01-25:
> I set to 0 as an initialization and this is an int.

In the python code you mean? Can you show how you're doing this?

raysalemi, 2021-01-25:
I start out initializing it all to 0.
```
    async def driver_bfm(self):
        self.dut.start = self.dut.A = self.dut.B = 0
        self.dut.op = 0
        while True:
            await FallingEdge(self.dut.clk)
            if self.dut.start == 0 and self.dut.done == 0:
                try:
                    (aa, bb, op) = self.driver_queue.get_nowait()
                    time.sleep(0.1)
                    self.dut.A = aa
                    self.dut.B = bb
                    self.dut.op = op
                    self.dut.start = 1
                except queue.Empty:
                    pass
            elif self.dut.start == 1:
                if self.dut.done.value == 1:
                    self.dut.start = 0
```
Then later I sometimes get an error that says `int` doesn't have `value` so I catch it.
```
    async def cmd_mon_bfm(self):
        prev_start = 0
        while True:
            await FallingEdge(self.dut.clk)
            try:
                start = int(self.dut.start.value)
            except ValueError:
                start = 0
            if start == 1 and prev_start == 0:
                self.cmd_mon_queue.put_nowait((int(self.dut.A), int(self.dut.B), int(self.dut.op)))
            prev_start = start
```

raysalemi, 2021-01-25:
Maybe the A = B = C = 0 construct doesn't do what I think it should?

eric-wieser, 2021-01-25:
> Then later I sometimes get an error that says int doesn't have value so I catch it.

Are you _sure_ that's what the error is? I'd guess it was `ValueError: invalid literal for int() with base 10: 'z'`, which is a very different problem - `.value` _does_ exist, but it contains `BinaryValue('z')`.

raysalemi, 2021-01-25:
> > Then later I sometimes get an error that says int doesn't have value so I catch it.
> 
> Are you _sure_ that's what the error is? I'd guess it was `ValueError: invalid literal for int() with base 10: 'z'`, which is a very different problem - `.value` _does_ exist, but it contains `BinaryValue('z')`.

I think you're right. I was wondering about that because trying to do 5.value() will return an AttributeError not ValueError

raysalemi, 2021-01-25:
I've seen both types of errors, but I think this question "How do I convert z to an integer" may be the real one.

eric-wieser, 2021-01-25:
In that case, I think your approach is already the right one - there's no "right" answer for choosing an integer that corresponds to `z`, so cocotb makes you the user choose with `try` / `except`. You could write your own function:
```python
def as_int_or_zero(value: cocotb.binary.BinaryValue):
    try:
        return int(value)
    except ValueError:
        return 0
```
```python
done = as_int_or_zero(self.dut.done.value)
```
which is slightly better because it ensures you only catch the valueerror caused by the X and Zs, and not other internal value errors which cocotb might produce while reading the value.

closed 2021-01-25 by raysalemi