cocotb/cocotb #2373: parameterizing test and passing argument via the command line

ktbarrett, 2021-01-25:
There was some discussion during the hackathon regarding the need to parameterize a test and feed argument from the command line. @alexforencich brought this up, and I too have need a feature like this. 

### Requirements
* Must support multiple tests per simulator run, like we do currently
* Should be able to pass arguments to particular tests
* Should be able to parameterize multiple tests with the same parameter
* Should have support as a make variable in the current flow

### Plusargs
Currently we have plusargs, which are also fed to the simulator. There is a limit to what can be passed in a plusargs, so this might not be the best answer. And officially they are only supported by SystemVerilog compatible simulators. GHDL, for instance, consumes them as simulator arguments. But they are supported in the make flow currently.

### Environment Variables
There is also ad-hoc argument passing via environment variables. This does work, but could be argued that it's tedium on the user that could be handled automatically. It's up to the user to decide how to handle each variable, and environment variables can be set in the current makefile flow.

### Pytest
OOTB pytest does not support this easily. According to [this](https://stackoverflow.com/questions/40880259/how-to-pass-arguments-in-pytest-by-command-line) answer it requires adding argument parser in conftest.py and then more code in the test to handle those arguments. This isn't really possible without extra work, due to inversion of control. And it requires the user to do most of the work because of the need to modify pytest's argument parser. The arguments would also need to be fed to the call to the main pytest invocation by way of environment variable. This ends up not being a very clean solution.

marlonjames, 2021-03-19:
If we were to use pytest as the regression manager, we could provide a general command line option that accepts some syntax for parameterizing tests, and apply them as appropriate. Similar to how the single `--tx` option in `pytest-xdist` takes multiple forms.

themperek, 2021-03-19:
Example pytest plugin I did: https://github.com/themperek/cocotb-test/blob/master/cocotb_test/plugin.py 
One can add extra options quite easily.

themperek, 2021-03-22:
One more comment. 
With pytest what you can do is:

```python3
@pytest.mark.parametrize("seed", range(6))
def test_param(seed):
    pass
```

One can run such a test from the command line:
```
pytest -k test_param[1]
```
But if you would like to run with `seed==10` it is not straight forward and needs some extra code.

ktbarrett, 2024-01-10:
`@cocotb.parametrize` could be made to generate tests with more interesting names than `my_test_001`.

PR #3748 (2024-03-02): Better parameterize names
Follow on from #3717. Changes how `cocotb.parameterize` names tests to make them more descriptive and easier to match with `TESTCASE`. Closes #2373.
closed 2024-03-02 by ktbarrett