cocotb/cocotb #2375: VHDL string writing has problems

cmarqu, 2021-01-25:
In https://github.com/cocotb/cocotb/pull/2329#commitcomment-46285975 it has been discovered that writing a string to a VHDL design does not end up in the string signal (although there is an  event or transaction on the signal so that a VHDL process is triggered).

The written string can be read back immediately however, that's why our existing simplistic testcases did not find this.

It's still unclear whether reading a string driven by the VHDL side can be read into cocotb correctly.

ktbarrett, 2021-01-25:
Is this with a particular simulator or does it affect all VHPI users?

cmarqu, 2021-01-25:
I tested Xcelium, Questa, and GHDL, all with the same behavior.

ktbarrett, 2021-01-25:
Odd, they all use different interfaces.

cmarqu, 2021-01-25:
Indeed, this may suggest that the value gets nulled before we hand it to the different interfaces. But how can we then read it back (I didn't try that explicitly with GHDL)? More investigation needed.

cmarqu, 2021-01-26:
Looks like I have just confused myself.

closed 2021-01-26 by cmarqu