cocotb/cocotb #2380: test_array_buses with Verilator breaks with "IndexError: in_valid(GPI_REGISTER)"

aignacio, 2021-01-29:
Hi, 
I'm trying to run the test case from Cocotb called test_array_buses and I'm seeing the following issue.  Is there an issue when using it with Verilator?

**Verilator version:** 4.106 2020-12-02
**Cocotb:** cocotb v1.5.0.dev243+g5d2d7c7c

```bash
0.00ns WARNING  cocotb.gpi                                GpiCommon.cpp:394  in gpi_get_handle_by_index         Failed to find a handle at index 0 via any registered implementation
     0.00ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: test_array_buses (result was IndexError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/Users/aignacio/projects/temp/test_array_buses.py", line 66, in test_array_buses
                                                                                                                         in_data_0 = TestDriver(dut, "in", dut.clk, array_idx=0)
                                                                                                                       File "/Users/aignacio/projects/temp/test_array_buses.py", line 19, in __init__
                                                                                                                         BusDriver.__init__(self, entity, name, clock, **kwargs)
                                                                                                                       File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/drivers/__init__.py", line 245, in __init__
                                                                                                                         self.bus = Bus(
                                                                                                                       File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/bus.py", line 85, in __init__
                                                                                                                         self._add_signal(attr_name, signame, array_idx)
                                                                                                                       File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/bus.py", line 104, in _add_signal
                                                                                                                         handle = handle[array_idx]
                                                                                                                       File "/usr/local/Cellar/python@3.9/3.9.1_6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/handle.py", line 594, in __getitem__
                                                                                                                         raise IndexError("%s contains no object at index %d" % (self._fullname, index))
                                                                                                                     IndexError: in_valid(GPI_REGISTER) contains no object at index 0
     0.00ns ERROR    cocotb.regression                         regression.py:478  in _log_test_summary               Failed 1 out of 1 tests (0 skipped)
     0.00ns INFO     cocotb.regression                         regression.py:551  in _log_test_summary               *******************************************************************************************
                                                                                                                     ** TEST                               PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     *******************************************************************************************
                                                                                                                     ** test_array_buses.test_array_buses    FAIL            0.00          0.00         0.78  **
                                                                                                                     *******************************************************************************************

     0.00ns INFO     cocotb.regression                         regression.py:568  in _log_sim_summary                *************************************************************************************
                                                                                                                     **                                 ERRORS : 1                                      **
                                                                                                                     *************************************************************************************
                                                                                                                     **                               SIM TIME : 0.00 NS                                **
                                                                                                                     **                              REAL TIME : 0.01 S                                 **
                                                                                                                     **                        SIM / REAL TIME : 0.17 NS/S                              **
                                                                                                                     *************************************************************************************

     0.00ns INFO     cocotb.regression                         regression.py:259  in tear_down                       Shutting down...
- :0: Verilog $finish
```

marlonjames, 2021-01-29:
I believe you are encountering this: https://github.com/verilator/verilator/issues/2727
There are several issues with Verilator's VPI in 4.106, and unfortunately 4.108 has a different problem that causes cocotb to not work.

There is ongoing effort to improve the Verilator VPI, but at this point Verilator + cocotb is unstable/experimental.

aignacio, 2021-01-29:
Hi Marlon,
thanks for the answer, is it better with ius/xcelium? what's the golden simulator for SV with Cocotb?

ktbarrett, 2021-01-29:
Seeing as that only commercial sims remain I don't think we are going to comment on the "best" one. We do test Riviera in CI, so take that for what it's worth.

aignacio, 2021-02-12:
Works with latest/greatest from master branch of verilator...

closed 2021-02-12 by aignacio