cocotb/cocotb #2381: Integrate with VUnit

ktbarrett, 2021-01-30:
Have you seen VUNIT [https://github.com/VUnit/vunit](url)?  This might be an interesting integration.  It even has file dependency processing.

_Originally posted by @leftink in https://github.com/cocotb/cocotb/issues/280#issuecomment-221286250_

[Some experimentation done](github.com/ktbarrett/vunit-cocotb). [Tracking in the VUnit repo](https://github.com/VUnit/vunit/issues/651).

jwprice100, 2021-02-09:
Is the long term plan to use vunit as a test runner for cocotb test cases? What are the timeframes here?

I've got a home grown cobbled together thing for work that gets the job done. It works well, but isn't pretty on the inside. I'd love to have something of a higher quality that isn't just me maintaining internally. Happy to contribute.

ktbarrett, 2021-02-09:
> Is the long term plan to use vunit as a test runner for cocotb test cases?

There are no definite plans yet. I wanted the ability to use cocotb other project automation systems in general, with no specific one in mind. VUnit was one interest, since it's the most mature project out there. Another maintainer was interested in edalize/fusesoc. cocotb is a part of [OSVB](https://umarcor.github.io/osvb/apis/core.html) and [PyFPGA](https://gitlab.com/rodrigomelo9/pyfpga) projects which have their own solutions. And there is interest in redesigning certain aspects of those stacks so that everything fits together nicer. Part of that effort is refactoring cocotb to be more of a collection of independent parts, and less of a monolith. That has been a long term goal of mine.

> What are the timeframes here?

The 2.0 refactor/rewrite is happening this year after the 1.5 release; so this likely won't be worked until after that time, unless we get more contributors. Developments in the FOSS HDL community might take a couple of years. It certainly won't be soon.

> I've got a home grown cobbled together thing for work that gets the job done

Yeah, me and @marlonjames do the same for our companies. I bet there are a dozen other solutions out there as well, all behind closed doors. It will be nice to have a high quality FOSS solution for project management and testing that support many methodologies, including cocotb. I think that is a major concern of the community right now, so it's definitely being worked.

> Happy to contribute.

If you are interested in contributing to cocotb in particular, you can start with [CONTRIBUTING.md](https://github.com/cocotb/cocotb/blob/master/CONTRIBUTING.md). Some important issues/PRs pertaining to bettering extensibility in cocotb are
* encapsulating and extending GPI implementation #2347, #1332
* makefile replacement #2381, #1562
* extend users for the GPI and Python GPI layers #1225, #1994
* supporting GPI-level extensions #89, #1217 

In terms of a community developed high-quality project automation, right now things aren't super organized. The best way to contribute is to be active in the gitter channels where this stuff is being talked about, and make your interest known. Pinging @umarcor.

umarcor, 2021-02-09:
Kaleb explained everything quite nicely already.

@jwprice100, if you are interested in combining VUnit and cocotb, and contributing with your knowledge/experience in that regard, please do read VUnit/vunit#651 (refed above). There, the philosophical differences and traditions of both user bases are discussed and a common ground is laid. This was Kaleb's prototyping: https://github.com/ktbarrett/vunit-cocotb. I forked that repo, and I implemented some alternative experiment. Then, I renamed the repo to OSVB. You can see that it's still shown as a fork of Kaleb's: https://github.com/umarcor/osvb. Precisely, https://github.com/umarcor/osvb/tree/master/SFF contains the vunit-cocotb sources, and a README explaining some details.

My approach is to preserve HDL testbenches as done in VUnit. Cocotb can drive declared but undriven signals in the testbench. The test finalization needs to be synchronized between VUnit and cocotb. I had some dialogues with Kaleb and Lars (VUnit) which clarified how it can be done. The example uses VUnit's private procedure `p_disable_simulation_exit(runner_state);` (https://github.com/umarcor/osvb/blob/master/SFF/test/tb_dff_cocotb.vhd#L27) for letting cocotb decide when to finalize. I need to update it with the opposite example (customization of cocotb for toggling a signal at the end).

Anyway, my approach is for allowing VUnit users to add cocotb into their workflow. Most Pythonic users of cocotb will find the requirement of having a top-level HDL testbench to be unacceptable for their workflows, because many cocotb features are not supported (it's just for co-simulating through Python using VPI/VHPI). Therefore, there is a lot of room for improvement. The strategy is for each project to clarify their own stack so we can then compare apples to apples, discuss, and converge towards common interfaces for e.g. defining sources, interacting with simulators, test/regression management, vendor log parsing/error reporting... That's a mid-long term goal. In RealWorld, we are tackling projects by pairs.

If you like my vunit-cocotb approach, with its constraints and advantages, that can be usable in a few weeks/months, if anyone can devote time to it. Unfortunately, none of the most active actors can prioritize it at the moment. A 1-2y frame is more likely unless someone steps ahead.

---

ghdl-cosim contains a dockerfile for testing vunit-cocotb together: https://github.com/ghdl/ghdl-cosim/blob/master/docker/vunit-cocotb.dockerfile. It's similar to the one in OSVB, but ghdl-cosim uses `master` branches and the latter uses submodules: https://github.com/umarcor/osvb/blob/master/.github/Dockerfile. In the very short term, I expect to have that image available through [hdl/containers](https://github.com/hdl/containers). GHDL is available already. I will push verilator in the following hours. Then iverilog. Last, I'll create hdlc/sim, including all three simulators together with Cocotb and VUnit. I hope that will foster experimentation and collaboration.

/cc FYI @Paebbels @LarsAsplund @rodrigomelo9

jwprice100, 2021-02-09:
Thank you both for the detailed explanations here and in the various other forums. Is there a consensus to use vunit as a test runner? I saw some back and forth in a few places about using edalize. I'm bias towards vunit as it's by far the most user friendly HDL test runner I know of. I also don't believe there's any core technical challenge, just philosophical and details. 

- What features of cocotb are not supported by having a top level HDL testbench? I do this from time to time (performance reasons or connecting other IP). It is certainly nice to not need to make a separate testbench file. However that's the only downside I am aware of. I recognize for vunit to drive, the testbench wrapper is necessary and doesn't bother me.
-That said I really like the idea of generating testbench wrappers automatically. Could VUnit's HDL compiler provide the information or would that require extending it?
- VUnit and Cocotb have different test discovery mechanisms. I didn't see discussion of where that was reconciled. From what I can tell in the proof of concept work, all cocotb tests are grouped under a single test case to vunit. I'd like to see each test cased treated/ran separately the way vunit traditionally does. This will be particularly power when using multiple processes to speed up test execution. Is there thought given on how that might be done?

ktbarrett, 2021-02-09:
> What features of cocotb are not supported by having a top level HDL testbench?

I think it's more of an issue of an inability to easily coordinate the two.

> VUnit and Cocotb have different test discovery mechanisms. I didn't see discussion of where that was reconciled. 

It hasn't been. I suggested overloading VUnit's testing internals to support discovery with cocotb. I don't think it was received well. I'm not happy with cocotb's regression system either. I think the current path forward is to not use VUnit in the end. VUnit doesn't support packaging tools as flexibly as FuseSoC does, it has to specially support each "package". VUnit doesn't support as wide of a variety of tools as cocotb or some hypothetical project automation system. VUnit has coupled test discovery mechanism into it's core so that in can really only support HDL-based tests, as you pointed. I think the final solution will only use pieces of VUnit, or something else altogether. Or maybe it will use VUnit =P We don't know yet.

umarcor, 2021-02-09:
> Is there a consensus to use vunit as a test runner? I saw some back and forth in a few places about using edalize. 

Not at all! Some users of cocotb do want to use VUnit, others want to use the simulator interface only, others don't want to use VUnit at all. As Kaleb said, the updates in cocotb 2.0 might allow to interface other projects more easily, one of which can be VUnit. But there is no consensus about using it. 

In the wider community, each project has a different and independent plumbing (OSVVM, UVVM, SVUnit...). That was my motivation for creating OSVB and pyCAPI (a very very simple proof of concept).

> I'm bias towards vunit as it's by far the most user friendly HDL test runner I know of.

That is my perception too, but I'm also very biased because I do want to have HDL testbenches. I like VUnit's approach focused on HDL designers, and allowing them/us to pick additional features non-intrusively. Yet, I know that this is my very personal preference. For OSVB, I'm willing to replace the VUnit module blocks in the main diagram with whatever comes out from the community. I don't have any strong feeling. Any improvement will be better than the currently very fragmented view for newcomers.

> I also don't believe there's any core technical challenge, just philosophical differences and details.

Do not dismiss philosophical differences and details. We are human. Concepts, motivations and ego can play a very relevant role in our behaviour. I don't mean it in any pejorative sense. It's difficult to tell someone that they should ditch part of the code they've worked on for several months/years. We need to build examples and show the benefits of each project, so that maintainers are naturally convinced about what is worth maintaining/improving.

NOTE: I co-maintain VUnit, but I did not write most of the codebase. Hence, it's not painful for me to stop using part of the codebase. But I'll only do so if I find better implementations. Meanwhile, it's the most complete/robust solution for simulation.

> * What features of cocotb are not supported by having a top level HDL testbench? I do this from time to time (performance reasons or connecting other IP). It is certainly nice to not need to make a [...]

The tests I did involved a single cocotb test. I ignore cocotb's multiple test management features, so I avoided them for now. That is related to logging/reporting. Cocotb is loaded dynamically by the VPI/VHPI passed as an argument to the simulator. Therefore, the Python instance where Cocotb is executed is completely independent from the one where VUnit is executed. In practice, VUnit and cocotb might be written in different languages and the result would be the same. That is good, because it allows cocotb to be started from any project management tool (the default are makefiles). Yet, it's undesirable because we cannot take direct advantage of VUnit and cocotb being written in the same language. Currently, cocotb reports using it's own mechanism, independent from VUnit.

Furthermore, my vision of cocotb is that it provides shared libraries with bindings to Python matching VPI/VHPI. I don't want test management features, and I don't want any complexity related to logging and generating XML (XUnit). More on this below. 

> * That said I really like the idea of generating testbench wrappers automatically. Could VUnit's HDL compiler provide the information or would that require extending it?

This is a concurrent collaboration area that is advancing slowly. VUnit has pretty decent VHDL and Verilog parsers for analizing the dependencies between files. But those are based on regular expressions. As a result, it has some limitations which are difficult to overcome. It is not detailed enough for parsing the ports of an entity, getting the direction and the types, and generating a testbench template where the entity is instantiated and connected to signals in the testbench.

Replacing VUnit's regexp based parser has been discussed before. Candidates were/are rust_hdl, pyVHDLParser and libghdl-py (GHDL). rust_hdl was not used because it is a compiled tool (Rust), so it would make distribution of VUnit more complex. Since then, Olof is not actively developing rust_hdl or VUnit. pyVHDLParser is based on Python, so it would fit really well. However, Patrick did not finish the development yet. He is very busy and can only work on it during holidays. Precisely, a couple of months ago, [vhdl/pyVHDLModel](https://github.com/vhdl/pyVHDLModel) was split from his parser, and GHDL's Python interfaces were overhauled for plugging them into pyVHDLModel. That's only a proof of concept for now. However, I believe that pyVHDLModel is the way to go. Users of GHDL will have the parser already, and users of vendor simulators might use pyVHDLParser or any other parser that produces a DOM.

Before the overhaul, I did some experiments with libghdl-py (GHDL) for implementing the very specific feature we are discussing here: https://github.com/ghdl/ghdl/pull/1449#issuecomment-692449255. So, it's possible to write a script in Python for using pyGHDL and getting a model of an entity and then generating a testbench. that script can be called from VUnit. Unfortunately, my example was broken when overhauling and I didn't have time to rewrite it based on the new `pyGHDL.dom`.

NOTE: getting the ports of an entity with pyGHDL.dom is also useful for [umarcor.github.io/hwstudio](https://umarcor.github.io/hwstudio/).

[... cut ...]
As you see, I'm not only proposing a top-level testbench to be required, but using the top-level HDL testbench as the main communication/coordination point between VUnit and cocotb. Following VUnit's philosophy, the HDL designer is in full control. This is very different from Kaleb's approach, which would be using VUnit's simulator interface as a replacement of makefiles, but avoid using the default VUnit runner (which is half Python half HDL) so that current cocotb projects can be used. My approach does require users to be actively willing to learn how to combine cocotb and HDL frameworks.

jwprice100, 2021-02-10:
> Do not dismiss philosophical differences and details. We are human. Concepts, motivations and ego can play a very relevant role in our behavior. I don't mean it in any pejorative sense. It's difficult to tell someone that they should ditch part of the code they've worked on for several months/years. We need to build examples and show the benefits of each project, so that maintainers are naturally convinced about what is worth maintaining/improving.

I do not mean to dismiss these as they are usually more important than the technical barriers. Indeed there are strong differences in opinions in all the various related threads of this topic! That appears to be one of the biggest obstacles to someone picking this work up to get done. 

> Furthermore, my vision of cocotb is that it provides shared libraries with bindings to Python matching VPI/VHPI. I don't want test management features, and I don't want any complexity related to logging and generating XML (XUnit). More on this below.

I completely agree. If Cocotb 2.0 is heading down a more modular approach, then I think this makes a ton of sense. It may even make sense for multiple projects to support cocotb. 

> Due to my simplistic projection of cocotb, I don't find useful to have multiple cocotb tests grouped together

I also agree. I prefer a single test to run at a time. This may because I had been using vunit for quite some time prior. There's also advantageous for parallel test execution. I have sims that sequentially take over 24 hours to run. So for CI it's really nice to distribute them across many CPUs. 

> Moreover, it would be possible to support having multiple tests defined in a single cocotb script, and have VUnit select which to execute. 

I like VUnit discovering multiple tests in a single cocotb module and selecting which test to run. I would like to combine that with VUnit's ability to runs sets of generic configurations to a single "hdl" test or now a cocotb module's tests. How would VUnit discover tests from a python file? My understanding is that it parses the HDL files to identify tests. It would be necessary to change vunit to support discovering tests from python modules which @ktbarrett seemed to indicate wasn't well received. It's obviously possible but does seem to clash with how VUnit design philosophy. I think this is where I get stuck the most.

> That would require some mechanism for the VHDL testbench to make a string available for cocotb to read. It would also require modifying cocotb for reading that parameter before starting an specific test.

You lose me a little here. Is the idea that the cocotb test name is passed in somehow (maybe as part of the runner cfg?) and then the setup functions would notify cocotb via some undetermined mechanism as to which test to run? 

In general I can see where the different philosophies clash here pretty hard. Is there no room for a middle ground? I have no idea what that might be though. I'm not strictly bothered by the HDL being in control. However keeping vunit as the master require a lot of work on the GPI backend which presents a non-trivial learning obstacle to someone like myself. I'm perfectly comfortable with C++ but neither the API that interacts with Python, nor the VPI/VHPI/FLI interfaces. That said I'm not unwilling to learn, the existing code in both code bases looks well organized.  I also don't like forcing one or two signals to exist that are only used by VUnit/Cocotb communication. That said I like that far more than makefiles.

@ktbarrett and @umarcor - I am willing over several months to figure out how to pull off vunit/cocotb since I'm comfortable with both and would really like an approach that I and my colleagues can use that isn't a cobbled together homegrown proprietary solution. I don't want to do anything not blessed by both communities though. Is that really the right thing for me to do? No matter the approach both repos would need changes to interact with each other.

rodrigomelo9, 2021-02-10:
Hi everybody. Thanks for the ping @umarcor. Let me say that I am a cocotb user (but I know nothing about internals) and I want to be a VUnit user, but not yet. I am developing pyFPGA (coming soon at https://github.com/pyFPGA/pyfpga) which is more related to synthesis stuff, but I am open to helping in the FOSS community around FPGAs. If somebody has clear what must be performed, let me know how to help :-D

umarcor, 2021-02-11:
> I do not mean to dismiss these as they are usually more important than the technical barriers. Indeed there are strong differences in opinions in all the various related threads of this topic! That appears to be one of the biggest obstacles to someone picking this work up to get done.

It is! Fortunately, Kaleb and me are in very good terms. We are both VHDL users, we have a very similar conception of how cosimulation interfaces can be used, and none of us has any economic interest in favouring any specific solution. The fait made our paths parallel and the worst case scenario is the current one: we working on the same features of isolated codebases. Therefore, we are together in the journey and we want to help each other fight against the technical difficulties. Just discussing our visions and stressing our differences is already very valuable for us to gain insight about our own ecosystems (which we both inherited from others, for good and bad). We "met" very few months ago, and this is not our main priority, so progress is slow. However, we contributed a lot in not-so-visible issues already: CI setup, simulation with GHDL, testing cocotb in containers, using cocotb on MSYS2... 

This is to say: join the trip! Let us know about your next stations, and we will try to make our routes cross. Communication in the open source (hardware) ecosystem is very distributed and sparse, but there is a lot going on underwater. Wanna dive in?

> I completely agree. If Cocotb 2.0 is heading down a more modular approach, then I think this makes a ton of sense. It may even make sense for multiple projects to support cocotb.

That is Kaleb's main objective. He does not necessarily want cocotb to be used with VUnit as I/we envision it. Yet, he wants to make it easier for other projects to use cocotb or for cocotb to use other projects (name it as you want :wink:). Hence, there will be opportunities, we just need to keep an eye on each other so we can spot them when they arise.

> I also agree. I prefer a single test to run at a time. This may because I had been using vunit for quite some time prior. There's also advantageous for parallel test execution. I have sims that sequentially take over 24 hours to run. So for CI it's really nice to distribute them across many CPUs.

Strictly, parallel test execution might be possible regardless of having tests defined together or separated. In VUnit, you can define multiple tests in a single testbench and then decide how to execute them. The same might apply to an integration with cocotb. However, I want to start with the most simple approach in order to address the most relevant technical issues first.

> I like VUnit discovering multiple tests in a single cocotb module and selecting which test to run. I would like to combine that with VUnit's ability to runs sets of generic configurations to a [...]

I was/am not thinking about VUnit discovering the list of available cocotb tests by parsing Python files and/or by importing the modules. My approach is direct: https://github.com/umarcor/osvb/blob/master/SFF/test/run.py#L18-L21. The MODULE is defined in the `run.py` file. By the same token, a `COCOTB_TEST` envvar could be defined (in fact, I wouldn't be suprised if cocotb supported it already). Therefore, you can use VUnit's ability for customising testbenches, HDL tests, and/or for generating configurations. You just need to add a parameter in the `run.py`, as you would do for waveform generation or any other init/do file.

VUnit discovering cocotb tests is only required for Kaleb's approach: using VUnit's simulator interface but forking/modifying the VUnit runner for removing the Python-HDL runner and replacing it with a cocotb-runner. My approach is very purposely constrained because I'm trying to preserve VUnit's and cocotb's codebases almost unmodified. I want to prove that both projects can already coexist with at most 20-30 lines contributed to each of them (if any). Once we are there, we can think about how to make both projects love each other.

Parsing/loading a cocotb module for discovering tests can be done in the `run.py` file, without modifying VUnit or cocotb. That code can be provided as a Python module, similarly to how pyCAPI is used in OSVB: https://github.com/umarcor/osvb/blob/master/AXI4Stream/test/vunit/run_capi.py.

> @ktbarrett seemed to indicate wasn't well received. It's obviously possible but does seem to clash with how VUnit design philosophy. I think this is where I get stuck the most.

This is open source. Back to technical vs philosophical conflicts. No matter what maintainers of a project think about your enhancements, if you believe there is good reason for trying a technical solution and you can test it, go ahead. Take you time, knowledge and other possible limitations into account; but dismiss any criticism which is not funded on technical constraints. Good maintainers will almost always reply to technically exciting questions, regardless of their opinion about your target. Naturally, this applies to myself too (and first).

In this specific case, Olof said that he would not like cocotb specific code to be added into the core. That's fair. Nothing prevents us from renaming VUnit/vunit to VUnit/core and creating VUnit/vunit. Or keeping VUnit/vunit as is and creating VUnit/bundle. Or keeping VUnit org as is, and using OSVB (which might be moved to hdl/osvb if there is community support for that). By the time we sort out the technical challeng [... cut ...]
I cannot promise any code to be accepted in any of the projects, because that depends on the specific code and on several other people. Anyway, I can tell that good solutions will be blessed regardless of the location of the code. I would suggest forking Kaleb's vunit-cocotb or umarcor/osvb and working there. VUnit and cocotb are added as submodules. In OSVB, CI builds a container with those submodules and runs tests in there. Therefore, you can set the submodules to your modified versions of VUnit and cocotb.

jwprice100, 2021-02-11:
Thank you for the response. I am very happy to help, but am just getting my feet wet with this. I appreciate all the insight provided thus far. I'm fairly passionate about making HDL simulation easier and I think VUnit and Cocotb both provide the best tools in the industry for this. I'd like to marry things up somehow, but I am not wedded to a particular solution. I fully agree no solution forward is going to be quick and am happy to plan longer term. I'm going to share my initial thoughts, please use your combined experiences to guide as you see fit.

So it seems it is desired that vunit not make any specific changes to accommodate cocotb and vice versa. Than _perhaps_ the best path forward is to create a project which extends both of them. Similar to what you both have started, but going the extra mile.  What I'm thinking in my mind is subclasses objects like vunit's testbench class to handle running a cocotb testbench vs. a vunit one. Then each test, vunit or cocotb could the master and the status/logging just has to be reconciled somehow. What would be critical in my mind is keeping the API essentially identical to VUnit's with some additions for adding/discovering cocotb modules and tests. The user interface should also be identical at this point.  This seems to be more or less what Kaleb was proposing, but instead of making those changes in the VUnit repo, do it in this yet to be named project. 

There are a lot of details for something like this to work. Yet I feel it's something that I could be reasonably accomplished. I've sort of done something like this already at work, which has given me insight in to how it could be done better. I'm sure certain changes to make certain aspects of VUnit and Cocotb more general would be required, but hopefully nothing specific to each other. 

Now this addresses really only one of your three use cases (replacing make files with the VUnit compiler/CLI).  It partially addresses the use case of the best of  VUnit and cocotb. Using cocotb to implement black boxes is interesting, but that makes cocotb more a slave then. I think that's ultimately doable in a project like this, but not something I would want to bite off at first.

umarcor, 2021-02-11:
Do not worry about the "black boxes" use case. All your concerns are legit and your preferences are equally valid. Build on top of Kaleb's proposal; I will support you within my knowledge.

> So it seems it is desired that vunit not make any specific changes to accommodate cocotb and vice versa.

Do not tell this to anyone (:wink:) but I hope that both VUnit and cocotb will accept some code in the future, and make it optional for users. Yet, let's consider the worst case scenario, where that does not happen.

/ccing @GlenNichols here, because he has also developed some in-house solutions for extending VUnit. Glen, I don't know whether you use cocotb, but you might find this discussion interesting anyway.

ktbarrett, 2021-02-11:
I agree that I do not think that any significant accommodations to support cocotb or VUnit in the other's main project is necessary or desired; they are philosophically at odds. "cocotb" is a methodology that defines tests as being written in Python. While "VUnit" is a methodology tests are written in an HDL language. "cocotb" is a methodology where multiple tests are run in a single instance, and test results are aggregated during simulation into a XUnit output file. While "VUnit" is a methodology where single tests are run per simulator instance, results are aggregated outside the simulation, and can be output in several formats. Installing cocotb should always provide tools to reflect that methodology, however the API may change; likewise with VUnit.

However, "cocotb" is also a collection of technologies that are useful outside the scope of the original, and make the use those pieces independently should be easier. Everywhere a difference of opinion in end use could occur (using Python vs Java, using cocotb's Regression Manager, writing tests in Python vs HDL, using makefiles, etc.), the project should be split and a well defined interface created; to accommodate the development of alternatives for each of those decisions. I would hope that other projects are willing to make changes to their project to split them up into well regulated pieces according to the same principal; which is essentially the Single Responsibility Principal. I would make cocotb a collection of pieces with the main package a metapackage with the "blessed" toolset. If other tools were split according to SRP, projects like pyFPGA and OSVB could simply be metapackages with a little binding code. That is the ideal world I'd like to see; and I think it would make maintainers of those stacks happier.

I think we need to think a little more about the mature technologies we have developed over nearly a decade, and less about our ideology. What that means to me is that most of the work would be done in the respective organizations: cocotb, VUnit, etc. Changes made would be transparent to them, but transformational for everyone else.

Some thought needs to go into defining those boundaries in each project, and how to encapsulate the behavior in each box. Doing the work to get VUnit and cocotb working together would 1. get something working soon for people looking for the best of both world. 2. Help maintainers on VUnit and cocotb see where they need to define those boundaries. Both alternatives would be good to follow, since eventually both would be good to support.

Using cocotb as a slave for modelling incomplete HDL entities is very useful. We do that at work, but we also run the main test from Python. I know you, @umarcor, seemed interested in using cocotb as a slave and thus your interest in getting cocotb to work with the DPI. That will require some serious thought, but that is something I think would be generally useful for cocotb to include in the main repo. One possible way I just thought of was using asynchronous queues to talk between cocotb and DPI since the DPI/VPI callback streams can't touch. And any other changes you think cocotb could do to better support working as a slave are welcome. I know the biggest offender is the regression manager. My first task after the 1.5 release is to work on #1225. With the ability to overload the Python entry point, you can avoid the regression manager altogether for your tests.

cmarqu, 2021-02-11:
> By the same token, a `COCOTB_TEST` envvar could be defined (in fact, I wouldn't be suprised if cocotb supported it already).

It's there already indeed: https://docs.cocotb.org/en/stable/building.html?envvar-TESTCASE#envvar-TESTCASE

jwprice100, 2021-02-12:
I'm starting to dig into how vunit works. Along with the proof of concept work @ktbarrett and @umarcor I'm pretty confident that I can do things like subclass VUnit's user interfaces, testbench objects and simulator interfaces in such a way that VUnit think it's running a normal testbench, but cocotb is invoked. Cocotb provides all the hooks I need (I'm already using them for what I use at my job). It'll be important to clean things up eventually (like changing the python entry point) but not strictly necessary to show what could be. 

I think my short term goal then is to continue what Kaleb started. It would prove to be a good launching point for a conversation and provide users like myself with an interim solution. It might ultimately go away but as long as it contributes to the eventual direction this all takes I'll be very happy.  I'll create some project (and now for the hardest part, naming it) and borrow heavily from the concepts started. 

A couple of notes.
- VUnit supports far fewer simulators than cocotb which will obviously limit support. Easy to handle gracefully.  
- I'll probably be forced to duplicate code from VUnit to avoid having to touch VUnit directly. 
- I'll probably only be able to test the free simulators, and ultimately Riviera Pro once I'm far enough along I can ask work for Permission. This leaves out modelsim and questa. It'll also leave out Active HDL. Is there ultimately an environment where I can access those simulators for regression tests? I'm not sure how the community deals with that particular issue. Otherwise I'll just take my best stab and let others fix it!

jwprice100, 2021-02-15:
Alright, I've got a very very basic proof of concept available here: [VCST](https://github.com/jwprice100/vcst). I believe I went down a very similar approach to what Kaleb had envisioned, except I did this without touching VUnit or cocotb's code at all. There are some weird things behind the scenes as a result of that. However the basic concept of what I envisioned can be seen there. Compilation & Dependencies, multi-processes, running all tests or a selected set of tests, etc. This does require using VCST to create a vunit UI object, but the API is identical. 

I claim, even in this imperfect form what I have demonstrates functionality that is widely useful. For anyone who cares to take a look, feedback is welcome. This is my first attempt at such a thing.

AlexLao512, 2023-03-12:
Hi @jwprice100 I just saw what you are working on. I have a use case that might make sense to throw into this discussion.

At the moment, at work we exclusively use VUnit to drive all our testing, large code base, really troublesome to move, VCST looks like what we want. 

At home however, I am developing an independent HDL code base, I am very comfortable using VUnit because that is what I have used for the last few years so I kind of want to stick with it... however there is one thing I would like to try in CocoTB, that is in some of the VUnit testcases, I can test Python code that I normally use to interact with an programmed FPGA board (register read/write mostly) in simulation instead. I haven't used CocoTB yet so I apologize if I have some of its concepts wrong but what I want to do is I want to call, from Python, some of the methods in my test classes written in pure SystemVerilog, these are things like AXI AW channel aw_write(), AXI W channel w_write(), etc. Once I can do this, basically huge portions of my Python code base can be tested against the RTL in simulation.

ktbarrett, 2023-03-12:
@AlexLao512 
> want to do is I want to call, from Python, some of the methods in my test classes written in pure SystemVerilog

Unfortunately this is not possible (right now?). See #1989.