cocotb/cocotb #2384: Cocotb New Version Not Working with Questa VHPI

raysalemi, 2021-02-02:
Hi All,

The most recent version of Questa (2021.1)  implements enough VHPI to work with Cocotb (as a first step.)  Our tests show that the beta version of cocotb fails with Questa 2021.1 while the current version works. 

The Questa team would like to know the next steps to take to help debug this issue. Could you suggest next steps? For example, if we were to push our modifications to cocotb to a branch, would someone be able to debug it with Questa 2021.1?

Thanks!
Ray

PS. For those who don't know: I am the Aerospace and Defense Solutions Manager for Siemens EDA DVT division (I'm in marketing.) We make Questa.

ktbarrett, 2021-02-02:
@raysalemi Rick White informed me that I (Kaleb Barrett of Raytheon Technologies) would be testing 2021.1 internally before its release. I helped the team get started running the build against our regressions last week, after which they were seeing numerous errors.

The first thing I'd like to do is add the Questa VHPI to our build. Previously, I think the build team was working with the Aldec-specific VHPI library. Simulator-specific hacks in each implementation of our GPI means that what works for them, might not work for you. A recent change may have caused the break. I can have a look if I can get access to the build.

ktbarrett, 2021-02-02:
I have news that the tool is in the final stages of the approval process and I will have it to test soon.

raysalemi, 2021-02-02:
So you can use the Raytheon build to test cocotb?

Do you need us to push a branch with our mods to the build scripts?

ktbarrett, 2021-02-02:
If you could. Thank you.

themperek, 2021-02-03:
Sorry for off-topic. Maybe you can get a license so we can include Questa in CI. This can be private as is done for Aldec now.

raysalemi, 2021-02-03:
> Sorry for off-topic. Maybe you can get a license so we can include Questa in CI. This can be private as is done for Aldec now.

We've been wondering if you had a license.  Let me look into it.

ktbarrett, 2021-02-03:
Neither us personally or our care-taking organization, the [FOSSi Foundation](http://www.fossi-foundation.org/), have a Questa license for use in our CI. It would __also__ be very useful for [building binary packages for distribution that support the FLI](https://github.com/cocotb/cocotb/issues/2318#issuecomment-763793708). Pinging @imphil.

raysalemi, 2021-02-03:
Could you tell me more about the FOSSi Foundation and its relationship to
Cocotb?  I also hear talk of a "Cocotb Consortium".  Does that exist and
what is it?  Thanks!

raysalemi, 2021-02-03:
> If you could. Thank you.

Hi Kaleb.  It turns out that we have made no changes that require pushing.  Raghu Ardeishar is going to send you the instructions to run the limited set of Cocotb examples with Questa 2021.1's VHPI support level. Just to be clear: we're still under development with this code but the 1.5 issues are stopping our progress so we'd appreciate any help.

ktbarrett, 2021-02-03:
The FOSSi Foundation, primarily Phillip Wagner (@imphil), took over the cocotb project in late 2018 after the original authors Chris Higgs (@chiggs) and Stuart Hodgson (@stuarthodgson) could no longer maintain cocotb. The FOSSi Foundation's role in the day to day of the cocotb is minimal. They provide legal guardianship and manage assets, like our private Azure Pipeline and the Aldec Active-HDL license we use on it. Since the change in ownership he has added new [maintainers](https://github.com/cocotb/cocotb/blob/master/CONTRIBUTING.md#maintainers), including myself. The "cocotb consortium" is probably just the aforementioned list of maintainers; there is no official organization of that name.

As soon as I get access I will update cocotb to be able to support the new interface, hopefully before the release of 1.5, which is scheduled soon. And I will provide him some additional documentation on testing that new GPI implementation.

raysalemi, 2021-02-03:
So then they would sign the agreements for a Questa license?

imphil, 2021-02-03:
@raysalemi Thanks for opening this issue. I followed up via email.

raysalemi, 2021-02-15:
@ktbarrett , this is the bug

ktbarrett, 2021-02-15:
Could you provide a traceback or error message of the bug in question?

raysalemi, 2021-02-15:
I believe you’ve been working with Raghu on this. It’s the same issue.

raysalemi, 2021-02-15:
Questa Sim 2021.1 runs all the example designs with 1.4. But it fails with
1.5.

I’m working the process to get the FOSSi foundation Questa licenses to use
for QA and debug. I believe Raghu was going to work with you.

imphil, 2023-01-23:
Questa now works with VHPI, closing this issue.

closed 2023-01-23 by imphil