cocotb/cocotb #2385: Cocotb with scapy cosimulation

silenuszhi, 2021-02-04:
Hi.
I've been using Cocotb with scapy for simulating a tcp stack.
Till now I've successfully run the ping/arp procedure.
It's TCP stack <=>MAC <=> cocotb <=> TAP device on wsl2 windows10 2004

This is the code I'm using for ping
`@cocotb.coroutine
async def rgmii_sfuncion(a, b, c):
    while True:
        data = a.recv_raw()
        crc = crc32(data[1])
        rawdata = b'\x55\x55\x55\x55\x55\x55\x55\xD5' + data[1]
        for i in range(4):
            temp = (crc >> (8*i)) & 0xFF
            rawdata += bytes([temp])
        await b.send(rawdata)
        await b.wait()
        rx_frame = await c.recv()
        #a.send(rx_frame.data[8:])
        a.outs.write(rx_frame.data[8:])`

It needs a packet send followed by a packet recv.  So ping and arp works fine. I think.

So when it comes to tcp, the problem is how can I recv a packet while send a packet concurrently.

I'm thinking about 
'@cocotb.coroutine
async def rgmii_sfuncion(a, b, c):
    while True:
        data = a.recv_raw()
        crc = crc32(data[1])
        rawdata = b'\x55\x55\x55\x55\x55\x55\x55\xD5' + data[1]
        for i in range(4):
            temp = (crc >> (8*i)) & 0xFF
            rawdata += bytes([temp])
        await b.send(rawdata)
        await b.wait()
        #rx_frame = await c.recv()
        #a.send(rx_frame.data[8:])
        #a.outs.write(rx_frame.data[8:])

@cocotb.coroutine
async def rgmii_rfuncion(a, b, c):
    while True:
        rx_frame = await c.recv()
        #a.send(rx_frame.data[8:])
        a.outs.write(rx_frame.data[8:])
'

Is that work?
I've tried the code before but no fortune.

But there may be a problem in the tcp stack.

ktbarrett, 2021-02-04:
If you need to run two pieces of code at the same time (in terms of simulation time, not wall time) you would use `cocotb.fork`. For example:

```python
async def send(...):
    # ...

async def recv(...):
    # ...

@cocotb.test()
async def test(dut):
    # run send task concurrently
    send_task = cocotb.fork(send(dut, ...))
    # run recv task concurrently
    recv_task = cocotb.fork(recv(dut, ...))
    # wait until both tasks finish
    await cocotb.triggers.Combine(send_task, recv_task)
```

> But there may be a problem in the tcp stack.

I would suggest `cocotb.triggers.with_timeout` to timeout in the case the TCP stack isn't working correctly, so you don't hang indefinitely waiting for a response.

silenuszhi, 2021-02-04:
> If you need to run two pieces of code at the same time (in terms of simulation time, not wall time) you would use `cocotb.fork`. For example:
> 
> ```python
> async def send(...):
>     # ...
> 
> async def recv(...):
>     # ...
> 
> @cocotb.test()
> async def test(dut):
>     # run send task concurrently
>     send_task = cocotb.fork(send(dut, ...))
>     # run recv task concurrently
>     recv_task = cocotb.fork(recv(dut, ...))
>     # wait until both tasks finish
>     await cocotb.triggers.Combine(send_task, recv_task)
> ```
> 
> > But there may be a problem in the tcp stack.
> 
> I would suggest `cocotb.triggers.with_timeout` to timeout in the case the TCP stack isn't working correctly, so you don't hang indefinitely waiting for a response.
> 
> P.S. You don't need the `cocotb.coroutine` decorators anymore.

Thanks ktbarrett.

The problem is my send and recv function is both while true loop. Because I need forever grab the TAP packets and push them into my verilog in recv function ; while fetch packets from my verilog then send them to TAP packets in send function.

ktbarrett, 2021-02-04:
> The problem is my send and recv function is both while true loop.

This is the point of `cocotb.fork`. It creates two concurrent tasks that run "at the same time". So two infinite loops will work fine.

> In async def recv. I use the scapy.recv_raw() which is a blocking function

Ahh, I see. `scapy.recv_raw` and the other scapy calls are threaded blocking calls. Wrap the functions that block in `cocotb.external` to turn them into an coroutine that you can `await` on. For example:

```python
data = await cocotb.external(a.recv_raw)()
```

themperek, 2021-02-04:
FYI: [verilog-ethernet](https://github.com/alexforencich/verilog-ethernet) uses cocotb and scapy :
https://github.com/alexforencich/verilog-ethernet/blob/8e1ad2eba6e684d91488e50e6f1796de15bf094d/tb/eth_axis_rx/test_eth_axis_rx.py
https://github.com/alexforencich/verilog-ethernet/blob/8e1ad2eba6e684d91488e50e6f1796de15bf094d/tb/eth_axis_tx/test_eth_axis_tx.py

silenuszhi, 2021-02-04:
> This is the point of `cocotb.fork`. It creates two concurrent tasks that run "at the same time". So two infinite loops will work fine.
>
>
> Ahh, I see. `scapy.recv_raw` and the other scapy calls are [...]
Seems that after I changed it to await external stuff. Python can't grab any packets.
I've uploaded the tb_tese.py

[tb_test.zip](https://github.com/cocotb/cocotb/files/5927109/tb_test.zip)

silenuszhi, 2021-02-04:
> FYI: [verilog-ethernet](https://github.com/alexforencich/verilog-ethernet) uses cocotb and scapy : [...]

Tks. I'm going to find if there is a solution there.

alexforencich, 2021-02-05:
Well, it uses scapy, but only for parsing and building packets, not for external communication.

silenuszhi, 2021-02-05:
Ahh. I get the problem. Need sometime to figure it out. Update later. Close for now.

closed 2021-02-05 by silenuszhi

silenuszhi, 2021-02-05:
> This is the point of `cocotb.fork`. It creates two concurrent tasks that run "at the same time". So two infinite loops will work fine.
>
>
> Ahh, I see. `scapy.recv_raw` and the other scapy calls are [...]

I'm a such noob for python. My mistake. After a surround both scapy read / write function with cocotb.external.
Now it works like a charm.

Thankyou all for help. @ktbarrett @alexforencich @themperek