cocotb/cocotb #2389: [Suggestion] Waves for each test

aignacio, 2021-02-06:
Hey guys,

dunno if this feature is already in place (and I couldn't figure it out how to use it) but it would be nice to parametrize a way to dump waves (vcd/fst) for each test rather than the whole regression itself. In industry, it's common to see regressions in UVM env generating in this way and it also helps to debug if only one of the tests failed.

Thanks

themperek, 2021-02-06:
It is probably out of the scope of cocotb itself. It would complicate makefiles that we would like to keep simple.

You can quite easily make it by adding a second top-level module that would dump waveforms to file.

You can have a look here: https://github.com/alexforencich/cocotbext-axi/blob/master/tests/axi/Makefile

There is some support for this in [cocotb-test](https://github.com/themperek/cocotb-test): https://github.com/themperek/cocotb-test/blob/9bd8e5a25f4b59b717e80702aeb619b81701def8/cocotb_test/simulator.py#L360-L375

aignacio, 2021-02-06:
Hey @themperek,

thanks for the reply, but by the link you sent, if I understood it correctly it points to gen waves for  **all test together in once** single fst/fsdb/vcd, I'm looking for generating per test basis

themperek, 2021-02-06:
You can change dumpfile name as you like.

aignacio, 2021-02-06:
But are you saying when using along with cocotb-test? Sorry, it's not clear to me bc it seems that you add that by compilation/elab time where it happens previous the runtime when scheduler is calling different set of tests..

themperek, 2021-02-07:
> Sorry, it's not clear to me bc it seems that you add that by compilation/elab time where it happens previous the runtime when scheduler is calling different set of tests..

If you mind test generated by `TestFactory` than there is probably no easy way to do this, but you can run one `TESTCASE` at a time.

aignacio, 2021-02-12:
Gotcha, I'm using your cocotb-test + testfactory and it's going good.

closed 2021-02-12 by aignacio

themperek, 2021-02-12:
BTW: You can use `TESTCASE` to select just single test to run.