cocotb/cocotb #2390: cocotb cannot find root handle

JaewonHur, 2021-02-08:
Using 
cocotb v1.4.0, 
verilator v4.034,
python 3.6.9
on
Ubuntu 18.04

Cocotb cannot find root handle.
Giving `Makefile`, `TraceGenTile.v`, `TraceGenTile_module.py` below.

Simple `make` twice will build the simulator. (after first exiting due to warnings)

I was able to find and run simulation on simple examples but the above case does not work.

[report.zip](https://github.com/cocotb/cocotb/files/5943428/report.zip)

themperek, 2021-02-08:
It looks like somethng in verilator. If I make top level an empty module cocotb start correctly.

JaewonHur, 2021-02-08:
Commenting out system tasks (`fwrite`, `fatal`) in top module's always block also works.

ktbarrett, 2021-02-08:
The only supported version of verilator is 4.106. Older ones and newer ones contain bugs in the VPI implementation.

JaewonHur, 2021-02-08:
It seems verilator v4.106 does not work either...?

ktbarrett, 2021-02-08:
Have you tried running with Icarus? Verilator's VPI support (and thus it's support for cocotb) is still very experimental and doesn't work reliably yet. Could you paste the error message you are getting?

marlonjames, 2021-02-08:
Check the generated `Vtop__Syms.cpp` file in `sim_build` and see if the top-level module is set to `VerilatedScope::SCOPE_MODULE`.
That was the issue when I saw this no root handle problem previously (It was caused by `$display` in an `always` block in the module).

JaewonHur, 2021-02-09:
The top level module is set to `VerilatedScope::SCOPE_OTHER`.

Here's the error message.
```
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Verilator version 4.106 2020-12-02
     0.00ns INFO     cocotb                                      __init__.py:209  in _initialise_testbench           Running tests with cocotb v1.4.0 from /home/jwhur/.local/lib/python3.6/site-packages/cocotb
     0.00ns INFO     cocotb                                      __init__.py:229  in _initialise_testbench           Seeding Python random module with 1612830117
     0.00ns INFO     cocotb.gpi                                  VpiImpl.cpp:442  in get_root_handle                 Nothing visible via VPI
     0.00ns ERROR    cocotb.gpi                                GpiCommon.cpp:288  in gpi_get_root_handle             No root handle found
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/__init__.py", line 240, in _initialise_testbench
    raise RuntimeError("Can not find root handle ({})".format(root_name))
RuntimeError: Can not find root handle (TraceGenTile)
     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:314  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:269  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

marlonjames, 2021-02-09:
@JaewonHur Then most likely the same underlying issue in Verilator is the culprit here. I would make an issue in the Verilator github project and see if they can help find the problem.

closed 2021-03-18 by github-actions[bot]

Saikiran196, 2024-02-06:
When I am trying to execute Frequency divider python code, I am also getting error like Couldn't find root handle