cocotb/cocotb #2397: Make codecov reporting more reliable

imphil, 2021-02-10:
We see rather flaky codecov reporting, and I've identified some issues with it.

- [x] clang and macos uploads fail. We can safely disable them. #2393
- [x] Private CI uploads are done with the Python uploader, which doesn't support Azure Pipelines. That leads to missing commit metadata. The bash uploader should do the trick. (https://github.com/cocotb/cocotb-private-ci/commit/f3c6fa188fb11319e08d4b718507855780b6a3d7)
- [ ] The GH Actions uploader doesn't tell which matrix job the coverage is from. Add that.

marlonjames, 2021-02-10:
Delaying codecov comment: https://docs.codecov.io/docs/pull-request-comments#after_n_builds

imphil, 2021-02-10:
Yeah, the PR is already waiting on my side :-)

imphil, 2021-02-15:
The most recent set of updates seems to do the trick for now. No Windows C/C++ coverage yet, that's tracked in #2406.

closed 2021-02-15 by imphil