cocotb/cocotb #2405: VHPI error with VHDL code and Rising/FallingEdge functions

SpaceDreams, 2021-02-12:
```
I’m getting a fatal error when running a basic testbench with VHDL under Cadence’s Xcelium simulator. Using this exact testbench with
an equivalent Verilog or SystemVerilog design works perfectly but the VHDL design fails with the following log: 
make results.xml
make[1]: Entering directory `/path/to/design/clocktests/tests'
set -o pipefail; \
MODULE=timertest TESTCASE= TOPLEVEL=Clocktimer GPI_EXTRA=cocotbvhpi_ius:cocotbvhpi_entry_point TOPLEVEL_LANG=vhdl/opt/cadence/xcelium/XCELIUM1809/tools.lnx86/bin/xrun -timescale 1ns/1ps \
"" -v93 -licqueue -64 -xmlibdirpath sim_build -plinowarn -top Clocktimer -loadvpi /home/whitece/.local/lib/python3.6/site-packages/cocotb/libs/libcocotbvpi_ius:vlog_startup_routines_bootstrapxrun(64): 18.09-s006: (c) Copyright 1995-2018 Cadence Design Systems, Inc.
Loading snapshot Clocktimer.clocktimer:sim .................... Done
-.--ns INFO cocotb.gpi ..mbed/gpi_embed.cpp:74 in set_program_name_in_venv-.--ns INFO cocotb.gpi ../gpi/GpiCommon.cpp:105 in gpi_print_registered_impl-.--ns INFO cocotb.gpi ../gpi/GpiCommon.cpp:105 in gpi_print_registered_implxcelium> run
REPORT/NOTE (time 0 FS) from process :$PROCESS_000 (architecture Clocktimer.Clocktimer:sim)
Hello World!
-.--ns INFO cocotb.gpi ..mbed/gpi_embed.cpp:244 in embed_sim_init0.00ns INFO Running on xmsim(64) version 18.09-s006
0.00ns INFO Running tests with cocotb v1.4.0 from /home/whitece/.local/lib/python3.6/site-packages/cocotb
0.00ns INFO Seeding Python random module with 1613087656
0.00ns INFO Nothing visible via VPI
0.00ns INFO Found test timertest.timertest
0.00ns INFO Running test 1/1: timertest
0.00ns INFO Starting test: "timertest"
Description: None
0.00ns ERROR VHPI ERROR: Registered callback isn't enabled! Got 1000
  0.00ns ERROR Failed to register a value change callback
1.00ns ERROR Test Failed: timertest (result was TriggerException)
Traceback (most recent call last):
File "/path/to/design/clocktests/tests/timertest.py", line 12, in timertest
await FallingEdge(dut.clk)
File "/home/whitece/.local/lib/python3.6/site-packages/cocotb/triggers.py", linereturn (yield self)
File "/home/whitece/.local/lib/python3.6/site-packages/cocotb/scheduler.py", linetrigger.prime(self.react)
File "/home/whitece/.local/lib/python3.6/site-packages/cocotb/triggers.py", lineraise TriggerException("Unable set up %s Trigger" % (str(self)))
cocotb.triggers.TriggerException: Unable set up FallingEdge(ModifiableObject(Clocktimer.clk))1.00ns ERROR Failed 1 out of 1 tests (0 skipped)
1.00ns INFO *****************************************************************************
** TEST PASS/FAIL SIM TIME(NS) REAL TIME(S) RATIO(NS/S) **
*****************************************************************************
** timertest.timertest FAIL 1.00 0.00 268.57 **
*****************************************************************************
  1.00ns INFO *************************************************************************************
** ERRORS : 1*************************************************************************************
** SIM TIME : 1.00 NS** REAL TIME : 0.01 S** SIM / REAL TIME : 77.82 NS/S*************************************************************************************
  1.00ns INFO Shutting down...
Simulation complete via $finish(1) at time 1 NS + 0

\
-access +rwc -createdebugdb -makelib Clocktimer /home/whitece/digital/designs/whitece/clocktests/tests/../hdl/*.vhd 2>&1 | tee sim_build/sim.log
Did not detect Python virtual environment. Using system-wide Python interpreter
VPI registered
VHPI registered
Python interpreter initialized and cocotb loaded!
123, in __await__
546, in _resume_coro_upon
323, in prime
Trigger
**
**
**
**
Here is the VHDL design: 
library ieee ;use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned.all;
 entity Clocktimer is
port( clk: in std_logic;
count: out std_logic_vector(3 downto 0));
end entity Clocktimer;
 architecture sim of Clocktimer is
begin
  process is
begin
  report "Hello World!";
wait;
  end process;
 end architecture;

The python testbench. 
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import FallingEdge
 @cocotb.test()
async def timertest(dut):
  # create the clock and start it running
clock = Clock(dut.clk, 100.0, units='ns')
cocotb.fork(clock.start())
   await FallingEdge(dut.clk) # <- this line fails
  # await Timer(200) # <- this line works fine
  dut._log.info("Running test!")
await FallingEdge(dut.clk)
dut._log.info("No Errors!")

and the Makefile: 
 PWD=$(shell pwd)
 VHDL_SOURCES += $(PWD)/../hdl/Clocktimer.vhd
 TOPLEVEL := Clocktimer
 MODULE := timertest
 TOPLEVEL_LANG ?= vhdl
 SIM ?= xcelium
 SIM_ARGS += -v93 
export COCOTB_REDUCED_LOG_FMT ?= 1
  # DO NOT DELETE THIS LINE
include $(shell cocotb-config --makefiles)/Makefile.sim

Any help with this would be appreciated

```

marlonjames, 2021-02-12:
Pinging @cmarqu

cmarqu, 2021-02-12:
This is a duplicate of #1076 which unfortunately does not have a solution.

closed 2021-02-12 by imphil