cocotb/cocotb #2413: Questa 2020.x and future versions drop support for +acc

sjalloq, 2021-02-18:
I've recently hit a problem where Questa was segfaulting when running a simulation with System Verilog code.  After providing a testcase to Mentor they've said the issue is down to the new optimisation engine present from 2020.x onwards.

I've not looked in detail at the new options but the following is a quote from Mentor:

> Questa R&D expert found that it is simulated (without the crash issue and I verified that) if you’ll had the following option to the Makefile (it is for QIS optimization flow which is the new one since 2020.x Questa versions that enables full vopt simulations while providing debug or visibility capabilities – and replaces the +acc option flow – please refer to the Questa User’s Manual document for more information about it):
> 
> VSIM_ARGS= -voptargs="-access=rw+/."
> 

I've confirmed that adding these as `SIM_ARGS` to my Makefile fixes the issue.  I didn't modify Makefile.questa to remove `+acc` as recommended but it seemed to work.  I've asked Mentor to clarify exactly what is required.

The segfault didn't occur initially when I started developing my testbench but appeared after some time.  So it's quite possible this issue isn't visible for everyone all the time but it looks like a change will be required to the Questa flow.

Shareef.

ktbarrett, 2021-02-18:
I guess the question is now whether to introduce version checking into the makefile so we can decide to use `+acc` or `-voptargs="-access=rw+/."` Or if we should just remove `+acc` and let the users set the correct access optimization flags. You can use pragmas to control visibility of signals on only certain signals, and let the rest be optimized out. Top-level signals are always accessible and maybe your test only touched them. Always specifying `+acc` may be unnecessarily slowing down cocotb sims on Questa. Thoughts @themperek?

ktbarrett, 2021-02-19:
@raysalemi Do you, or would anyone at Siemens, have an opinion?

raysalemi, 2021-02-19:
> @raysalemi Do you, or would anyone at Siemens, have an opinion? We tag Questa issues with `category:simulators:questa` which you can use to keep up with Questa-cocotb issues

Hi @ktbarrett and @sjalloq,

I'd love to track these issues.  How do I subscribe to a categorization?

To answer your broader question Siemens EDA is transitioning to a new optimization system. I'm not sure of the details of interaction between our latest version 2021.1 and +acc.  I would ask your Siemens AE for those details.

sjalloq, 2021-02-22:
I've had some more feedback regarding the issue, it seems that until they fully drop support for `+acc` the two options can be used together. QIS doesn't support WLF dumping so there will be required changes to the way we dump waves.

For now, this seems to work:

```makefile
SIM_OPTIONS ?= -voptargs="-access=rw+/. +acc" -suppress 12130 
```

However, I'm also seeing other issue where the tool just isn't as stable as it used to be.  The driver tracing, for example, is pretty flaky.

IvanGrafsky, 2025-04-18:
Hi all!
I spent about week investigating crash of questasim and found this topic.
So I got something like
`# ** Error: (vopt-191) Questa has encountered an unexpected internal error: ../../src/velab/gbe_instbody.c(5109). 
# Please contact Questa support at http://supportnet.mentor.com/`
And I started my investigation with simplifying code and I got that if you want to instance interface with parameter and to fill this value from another instance of interface, like: 
`axi_if#(.DATA_W(axi_in.DATA_W)) int_axi()` and this leads to crash of questa with this error
so I changed Makefile.questa and removed COMPILE_ARGS += +acc - so no +acc in vlog and everything becomes fine.
Hope this helps