cocotb/cocotb #2415: Relese cocotb 1.5.0

imphil, 2021-02-20:
Milestone: https://github.com/cocotb/cocotb/milestone/15

Release process checklist:
- [ ] Close all issues blocking the milestone
- [ ] prepare RC release
- [ ] prepare release announcement email
- [ ] tag release
- [ ] test release build on PyPi
- [ ] send out announcement to mailing list

imphil, 2021-02-20:
I'm planning to do 1.5.0rc1 on Monday, please either get the remaining docs PRs in by then, or they'll need to wait until the next release. 
CC @cmarqu

ktbarrett, 2021-02-21:
Don't forget to run towncrier!

cmarqu, 2021-02-21:
https://github.com/cocotb/cocotb/wiki/Release-Process - needs a bit of clean up, but towncrier is listed.

imphil, 2021-03-04:
I just pushed rc2 containing a single fix, let's give it a week until March 11 to see if something major comes up and do the release otherwise.

ktbarrett, 2021-03-04:
Guess it's too late to merge #2426?

imphil, 2021-03-04:
No problem at all, RCs are cheap. Merge it, and I'll do another one tomorrow.

imphil, 2021-03-12:
1.5.0 is out now.

closed 2021-03-12 by imphil