cocotb/cocotb #2416: Use cocotb with the Xilinx Vivado Simulator

Reisswolf, 2021-02-21:
I am wondering if it could be possible to use cocotb with the Xilinx Vivado Simulator. The main reason this is interesting is because it would allow the use the encrypted Xilinx SecureIP in the source projects. I know that most of the commercial simulators support SecureIP but I was looking for a more affordable solution. 
Was this looked into already?

ktbarrett, 2021-02-21:
The Vivado simulator doesn't support a cosimulation interface like the VPI or VHPI, so cocotb support is not currently possible.

Reisswolf, 2021-02-21:
I have not looked through it carefully but the Xilinx Simulator does seem to have some external API that is called Xilinx Simulator Interface (XSI).

The general description from UG900 in Appendix K (page 252):
- "The Xilinx Simulator Interface (XSI) is a C/C++ application programming interface (API) to the Xilinx Vivado simulator (xsim) that enables a C/C++ program to serve as the test bench for a HDL design. Using XSI, the C/C++ program controls the activity of the Vivado simulator which hosts the HDL design."
- https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug900-vivado-logic-simulation.pdf

The XSI does not seem to follow the VPI or VHPI standard but it may or maybe not be usable as interface.

imphil, 2021-02-21:
XSI isn't unfortunately even close to being a simulator interface as needed for cocotb. For example, only top-level ports can be accessed. Supporting xsim in cocotb therefore isn't possible until Xilinx adds a VPI or VHPI interface to its simulator.

Until that happens please use another simulator, there are plenty to choose from :-)

closed 2021-02-21 by imphil

bp-ouhaha, 2023-12-07:
it sounds like DPI exists in Vivado simulator nowadays:
https://docs.xilinx.com/r/en-US/ug900-vivado-logic-simulation/Direct-Programming-Interface-DPI-in-Vivado-Simulator

ktbarrett, 2023-12-07:
DPI is not supported by cocotb, nor really could it be.

themperek, 2024-01-23:
@Reisswolf You can try https://github.com/themperek/cocotb-vivado. It is early version and  very limited because of limited XSI interface.