cocotb/cocotb #2418: pyuvm: The Python version of the UVM

raysalemi, 2021-02-21:
Hi All,

I wanted to let you know that I'll be presenting a paper at DVCON named _Verification Learns a New Language: An implementation of IEEE 1800.2_, in which I'll present a personal project: I wrote the UVM in Python.

I released version 1.0 of **pyuvm** today on pip. The project is also [on Github](https://github.com/pyuvm/pyuvm) as an open source project.

I use **cocotb** in the example of **pyuvm** running on the Icarus simulator. (On my Mac!)

I'll also be publishing a book in April named _Python for UVM Verification_. The book teaches SystemVerilog engineers a little bit of Python. Then it teaches the UVM using Python (similar to the _UVM Primer_) and finally it shows the reader how to connect **pyuvm** to **cocotb**. (The **cocotb** section is short and teaches just enough to get the reader writing their own signal-level coroutines.)

One can install pyuvm with `pip`:

`% pip install pyuvm`

I thought you'd like to know about the project before it gets announced at large.

Paebbels, 2021-02-21:
@raysalemi looks like your project is not public ...  
I also had a look at [PyPI](https://pypi.org/project/pyuvm/). The used description text is completely broken.

raysalemi, 2021-02-21:
@Paebbels Thanks!  I fixed the visibility.  I had it private until I reached 1.0.

It looks like the description text doesn't parse Markdown tables. So, that will need fixing.  Thank you.

raysalemi, 2021-02-21:
> @raysalemi looks like your project is not public ...
> I also had a look at [PyPI](https://pypi.org/project/pyuvm/). The used description text is completely broken.
> 
> How does pyuvm compare to https://github.com/tpoikela/uvm-python ?

Hmm...I don't know. I didn't know about uvm-python.

sjalloq, 2021-02-21:
@raysalemi https://bit.ly/3dz8iwn

raysalemi, 2021-02-21:
> @raysalemi https://bit.ly/3dz8iwn

Hah!  Never thought to do it.

themperek, 2021-02-22:
@raysalemi Maybe you could already share an opinion on how does it feel with python compare to SV?

raysalemi, 2021-02-22:
> @raysalemi Maybe you could already share an opinion on how does it feel with python compare to SV?

The UVM is much easier to learn and use in Python than SV because there is no syntactic typing, and thus no parameterization.

For example you can create a component using the factory like this.

```python
class MyComponent(uvm_component):
   ...

class MyEnv(uvm_env):
   def build_phase():
        comp = MyComponent.create("comp", self)
```

We don't need the long incantation. The same applies to `ConfigDB().get(self,"","LABEL")`  Also components in the pyuvm are automatically registered with the factory, so no macros.

I refactored the phasing system so no `phase` variable.  (Also no custom phases that I've never seen used.)

raysalemi, 2021-02-22:
**pyuvm** also uses Python logging. So no UVM reporting system. (though there is component support for controlling logging hierarchically.)

tpoikela, 2021-02-23:
Nice work! Great to see that Python/cocotb is getting more attention. 

uvm-python is a side-project, I wanted to see if the library can be ported to Python using cocotb. It seems to work quite well, but the porting brings a lot of baggage from SV-side, and probably makes the code more complex than necessary. Some issues have also arisen due to SV UVM using zero delays and non-blocking region, which did not function exactly as expected in cocotb.

As mentioned in the comments of `pyuvm` code, many features like sequences and UVM phasing can be implemented in a simpler way. On the other hand, the user does not have to be aware of these underlying complexities (like they don't have to know about them on the SV UVM).

@raysalemi Is there a plan to include UVM registers in `pyuvm`?

raysalemi, 2021-02-23:
> Nice work! Great to see that Python/cocotb is getting more attention. 
> 
> 
> 
> @raysalemi Is there a plan to include UVM registers in `pyuvm`?

@tpoikela , thanks!

I've always imagined that registers would be one of the earliest pull requests. 

My biggest challenge going forward is to figure out how to implement the excellent testing system that **cocotb** has. I have tests that run with nosetest and one **cocotb** example, but I think I need to learn tox.

themperek, 2021-02-23:
>My biggest challenge going forward is to figure out how to implement the excellent testing system that cocotb has. I have tests that run with nosetest and one cocotb example, but I think I need to learn tox.

Maybe this can help? https://github.com/themperek/cocotb-test 
Some info: https://github.com/cocotb/cocotb/files/3908378/cocotb-test_WOSDV2019.pdf

raysalemi, 2021-02-23:
> Maybe this can help? https://github.com/themperek/cocotb-test
>
> Some info: https://github.com/cocotb/cocotb/files/3908378/cocotb-test_WOSDV2019.pdf

This looks great!

raysalemi, 2021-03-18:
> @raysalemi looks like your project is not public ...
> I also had a look at [PyPI](https://pypi.org/project/pyuvm/). The used description text is completely broken.
> 
> How does pyuvm compare to https://github.com/tpoikela/uvm-python ?

I've got a better answer now. uvm-python was generated by a script that converted SystemVerilog to Python.

I wrote pyuvm from scratch using the IEEE 1800.2 spec and refactoring the code to take advantage of Python.

closed 2021-04-14 by ktbarrett