cocotb/cocotb #2427: Problem during installation on Ubuntu 20.04

scraich, 2021-02-24:
Hi everyone, 
I encountered a problem while trying to install cocotb v1.5 on my computer both by cloning the official repo and from pypi, with:
- Ubuntu 20.04 
- Python version is 3.8.5 
- GHDL version is 1.0.0.
I have googled for a solution and tried out various fixes but nothing helped.
The error is the following:
``` 
  ...
  creating build/temp.linux-x86_64-3.8/cocotb/home/andrea/software/cocotb/cocotb/share/lib
  creating build/temp.linux-x86_64-3.8/cocotb/home/andrea/software/cocotb/cocotb/share/lib/gpi_log
  x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -I../../home/andrea/software/cocotb/cocotb/share/include -I/usr/include/python3.8 -c ../../home/andrea/software/cocotb/cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.linux-x86_64-3.8/cocotb/libs/libgpilog/../../home/andrea/software/cocotb/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -flto
  Assembler messages:
  Fatal error: can't create build/temp.linux-x86_64-3.8/cocotb/libs/libgpilog/../../home/andrea/software/cocotb/cocotb/share/lib/gpi_log/gpi_logging.o: No such file or directory
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly
```
Does anyone know how I could resolve this? 
Any help or suggestion would be appreciated!
Thank you

ktbarrett, 2021-02-24:
I just tried to replicate the issue in a clean docker image and did not see a problem. I have seen the error before when trying to compile on a RHEL 7 machine using an older version of a conda distribution, but couldn't duplicate it elsewhere. An editable install worked for me in a pinch, can you try that? `pip install -e .` in the local clone or `pip install -e git+https://github.com/cocotb/cocotb@stable/1.5`.

scraich, 2021-02-24:
I have just resolved by specifying also `--prefix` , as it would not work with just `-e`, nor with `--user`, giving an error as it would try to install in some directory with no permission. 
It is some problem on my machine, as a colleague of mine also did not have similar problems.  
Thank you very much for the hint, it lead to the solving.

closed 2021-02-24 by scraich