cocotb/cocotb #2431: List unsupported simulators and explain why they are not supported

cmarqu, 2021-02-26:
Every once in a while, the question about unsupported simulators, e.g. XSim, comes up.
Maybe we should add an "Unsupported Simulators" section under "Simulator Support" and explain why those are unsupported.

ktbarrett, 2021-02-26:
Should also include Xyce. I remember there being a discussion about that in some issue. While using Xyce with a typical simulator (say Icarus) with cocotb independently, *or* using cocotb to coordinate the two, is possible; directly using cocotb with Xyce is not possible.

cmarqu, 2021-02-27:
Another candidate: https://metrics.ca/

Lotion-3, 2026-11-16:
I'd like to take this. Could you assign it to me?

I have a change ready that adds an "Unsupported Simulators" section to the simulator support page, covering XSim and Xyce, based on what was said in #2416, #2727, #4601 and #2110. I left out Metrics DSim, since it's now supported as Siemens DSim.

cmarqu, 2026-11-16:
These days, the landscape of simulators is quite different, so we shouldn't list individual simulators, but rather point out that we do not add special support for new simulators, and why.

See e.g. https://github.com/cocotb/cocotb/pull/5795#issuecomment-5776313371 and https://github.com/cocotb/cocotb/issues/5918#issuecomment-6773036390

ktbarrett, 2026-11-16:
It would be nice to figure out how we plan on supporting these simulators at the GPI level. Most simulators do a pretty poor job of following the spec (also the spec isn't great), so we have to have "flavors." Ideally we could provide a "flavourless" build of the GPI for these simulators to use. It's just that we don't have any way to simulate that now as all current simulators have some "flavoring." And we should ideally support ways to load other GPI impls out-of-tree, but I haven't thought enough about that.

cmarqu, 2026-11-16:
> It's just that we don't have any way to simulate that now as all current simulators have some "flavoring."

We'll just have to vibecode our own flavorless simulator then ;)