cocotb/cocotb #2432: can cocotb help maintain the historical SV verification platform

leevic-star, 2021-03-01:
Hi everyone，I know cocotb for a short time, I want to know can cocotb help maintain the historical SV verification platform？
Thank you.

themperek, 2021-03-01:
> can cocotb help maintain the historical SV verification platform？

I think more information is needed. What exactly are you trying to do?

leevic-star, 2021-03-02:
> > can cocotb help maintain the historical SV verification platform？
> 
> I think more information is needed. What exactly are you trying to do?

I mean, whether cocotb supports interacting with SV or whether it is possible to use cocotb on the SV verification platform. If possible, the historical SV will be able to better transition to cocotb.

ktbarrett, 2021-03-02:
> whether cocotb supports interacting with SV

Yes, you can interact with SystemVerilog code by setting, reading, and reacting to value changes on signals and variables.

> whether it is possible to use cocotb on the SV verification platform.

"the SV verification platform"? There are many SystemVerilog verification frameworks, like UVM and SVUnit. Many of those frameworks were not ever meant to interact with a cosimulation framework, so I doubt meaningful integration can occur, but perhaps they can. 

> If possible, the historical SV will be able to better transition to cocotb.

I don't know anything about your set up, so I can't say whether it will be easy or not. You will need to provide more specifics if you want good answers. In broad strokes: yes, cocotb can do everything a SystemVerilog verification framework can, but I doubt integration will happen. The best way forward would be to simply not reuse your existing SystemVerilog framework and write new tests in cocotb. 

If you are accustomed to UVM, you can look at [uvm-python](https://github.com/tpoikela/uvm-python) or [pyuvm](https://github.com/pyuvm/pyuvm).

leevic-star, 2021-03-02:
> Yes, you can interact with SystemVerilog code by setting, reading, and reacting to value changes on signals and variables.
>
>
> "the SV verification platform"? There are many SystemVerilog verification [...]

leevic-star, 2021-03-02:
> 
> If you are accustomed to UVM, you can look at [uvm-python](https://github.com/tpoikela/uvm-python) or [pyuvm](https://github.com/pyuvm/pyuvm).

Thank you very much for your patient answer. I think I need to know more about cocotb, and pyuvm and uvm-python look very good. In fact, I want to reuse the previously developed verification IP (based on UVM verification frameworks) in cocotb, rather than redeveloping it...maybe it really  be better to write a new VIP in cocotb.

ktbarrett, 2021-03-02:
If you are interested in continuing to use UVM, just in Python. Talk to @tpoikela, he used scripts to translate UVM from SV into Python and then touched it up. Perhaps those scripts will work on your source as well?

tpoikela, 2021-03-02:
@leevic-star For sv2py porting, you can try https://github.com/tpoikela/uvm-python/blob/master/bin/sv2py.pl. However, if you choose to use it, you also need to use the full library https://github.com/tpoikela/uvm-python. This is a direct port of SystemVerilog UVM to Python, so the API is very similar to the original UVM. Some SystemVerilog built-in features must be converted manually though.

leevic-star, 2021-03-03:
@tpoikela @ktbarrett Thank you again,I'll try this.

raysalemi, 2021-03-11:
> Thank you very much for your patient answer. I think I need to know more about cocotb, and pyuvm and uvm-python look very good. In fact, I want to reuse the previously developed verification IP [...]

pyuvm is a little bit more Pythonic that I used Python logging and simplified phasing.

raysalemi, 2021-03-11:
> If you are interested in continuing to use UVM, just in Python, talk to @tpoikela. He used scripts to translate UVM from SV into Python and then touched it up. Perhaps those scripts will work on your source as well?

Scripts!  And here I am writing pyuvm by hand like a caveman.  On the other hand it gave me a chance to refactor portions.

closed 2021-04-19 by github-actions[bot]