cocotb/cocotb #2433: Cannot select testcase from the module that is not first in the MODULE list

sebastiancieslak, 2021-03-01:
Hi All,

test names passed by the TESTCASE attribute are only searched in the first module on the MODULE list. If they are not there, an AttributeError is raised, even if the tests are in the next modules on the MODULE list.

**Example:**
_Makefile:_ MODULE = x_tests,y_tests
_x_tests.py:_ x_test
_y_tests.py:_ y_test

make sim -> 2 tests runned
make sim TESTCASE=x_test -> 1 test runned
make sim TESTCASE=y_test -> AttributeError: Test doesn't exist in x_tests

I've already fixed this (changes in _discover_tests in regression.py), but due to your Contribution Guidelines decided to start with creating this issue first. I will open a pull request with the MRFE test in a moment.

ktbarrett, 2021-03-01:
Yeah, there is [a comment in the source stating as much](https://github.com/cocotb/cocotb/blob/9fe131d93f31438d032307f70096fac797af78e0/cocotb/regression.py#L204), but I'm not sure exactly why this decision was made. Either way, I agree the current behavior is poor.

ktbarrett, 2021-03-01:
As a quick fix, I think the logic can be modified so it selects the first test that matches the `TESTCASE` as it loops through the `MODULE`s. Maybe you can think of something better.

Requirements:
1. All `MODULE`s specified should be loaded *exactly once*.
2. Only the first test that matches `TESTCASE` (in order of appearance in `MODULE`) should be run.

x74ea6f, 2021-03-25:
Just idea. Is it possible to use regular expressions for TESTCASE?

I tried.

regression.py _discover_tests()
```python
            if test_str:
                # Specific functions specified, don't auto-discover
                test_inst_list = []
                for d in dir(module):
                    item = getattr(module, d)
                    if isinstance(item, Test):
                        test_inst_list.append(item.__name__)

                test_list = []
                for t_str in test_str.rsplit(','):
                    for inst in test_inst_list:
                        m = re.fullmatch(t_str, inst)
                        if m:
                            test_name = m.group()
                            if not test_name in test_list:
                                test_list.append(test_name)

                for test_name in test_list:
                    test = getattr(module, test_name)
                    test.skip = False
                    yield test

                # only look in first module for all functions and don't complain if all functions are not found
                break

            # auto-discover
```

ktbarrett, 2021-03-25:
@x74ea6f This was already resolved in #2434.

closed 2021-03-25 by ktbarrett