cocotb/cocotb #2435: Can ANSI avoid using BLUE_FG? Save my eyes please.

dukelec, 2021-03-02:
The default color:
[image]

After I changed the BLUE_xx in COLOR_INFO and COLOR_TEST in ANSI.py to BRIGHT_BLUE_xx and the result is much better:
[image]

By the way, thanks to the contributors of cocotb.

eric-wieser, 2021-03-02:
Isn't this just a configuration issue with your terminal? If `BLUE_FG` is unreadable, then change that color to be readable in your terminal config!

eric-wieser, 2021-03-02:
For reference, the color pallete default across a selection of platforms:

https://en.wikipedia.org/wiki/ANSI_escape_code#3-bit_and_4-bit

ktbarrett, 2021-03-02:
The crappy dark blue color is default on many terminals. It's really of no consequence what the color is. Bright blue will be better for the majority of people without additional configuration, and it's a 1 line change.

dukelec, 2021-03-02:
As a Linux user for more than 10 years, I didn't even know that I could set the color of the terminal settings menu.
I am using Gentoo system, terminal xfce4-terminal 0.8.9.2, the first screenshot is the default color "White on Black".

marlonjames, 2021-03-12:
Should we leave the `Running test x/y:` with a blue background or change to white?

ktbarrett, 2021-03-12:
I noticed that as well. I think marking test start and test end help visually parse the output. We aren't marking test end though.