cocotb/cocotb #2437: Compiling for Verilog instead of SystemVerilog on cocotb

kgokarn, 2021-03-02:
Cocotb version: .4.0
OS: Ubuntu 18.04LTS 64-bit
Python: 3.6 System
Simulator : Icarus Verilog 11.0

I want to know if cocotb makefile can be edited in a way that the hardware is compiled for Verilog HDL rather than SystemVerilog.

Thanks in advance.

themperek, 2021-03-02:
>I want to know if cocotb makefile can be edited in a way that the hardware is compiled for Verilog HDL rather than SystemVerilog.

More info needed.

ktbarrett, 2021-03-02:
This is up to the simulator. You can select the language version using the `-gXXXX` option. See [here](https://iverilog.fandom.com/wiki/Iverilog_Flags#-g.3Cgeneration_flag.3E). You can pass this to the simulator with cocotb using the [`SIM_ARGS`](https://docs.cocotb.org/en/stable/building.html#var-SIM_ARGS) and [`COMPILE_ARGS`](https://docs.cocotb.org/en/stable/building.html#var-COMPILE_ARGS) make variables.

If you have any other simulator-specific questions, take them to the appropriate place.

closed 2021-03-02 by ktbarrett