cocotb/cocotb #2440: Testbench hangs indefinitely on v1.5

aignacio, 2021-03-02:
Hi everyone,

I'm trying to run a simple testbench for the following project but I don't know why the simulator just hang indefinitely  and it's hard to debug the bkg reason. I'm using xcelium as the main simulator but I've tried with Verilator latest (4.110) and the behaviour is the same, is there a wrong setup in my tb class? In the Verilator run I can see that basically it hangs waiting for the clock events on ClockCycles. Tks ahead for the help!

https://github.com/aignacio/ravenoc/blob/master/tb/test_ravenoc_basic.py#L47

Verilator = 4.110
Xcelium = 18.03
Cocotb = 1.5.0rc2.dev1+gbd98f994

ktbarrett, 2021-03-03:
First thing: Only Verilator 4.106 works, newer version introduce a bug in the VPI that prevents cocotb from running correctly.

Second thing I see is in your TB you are referencing `RST_CYCLES`, which doesn't exist. You have a top-level HDL process which continues to run after cocotb dies because the startup path doesn't call `sim_end` if there is a failure (#2018).

aignacio, 2021-03-03:
Hey @ktbarrett 
thanks for the answer, the definition of RST_CYCLES is available [here](https://github.com/aignacio/ravenoc/blob/master/tb/default_values.py#L7) and it's imported in the main tb.

ktbarrett, 2021-03-03:
Ah, I didn't look too hard, I ctrl-f's for it and didn't see it. Could you switch to Verilator 4.106 and run it again to see if that's your only problem? If not, could you post where it was hanging?

aignacio, 2021-03-03:
On verilator 4.106 it seems to be working, thanks @ktbarrett. Is there a fix for Verilator on master branch?

closed 2021-03-03 by aignacio

marlonjames, 2021-03-03:
Verilator needs a fix for a VPI issue that makes cocotb incompatible: https://github.com/verilator/verilator/issues/2778