cocotb/cocotb #2456: looking for a feature to support .f filelist used with simulators like xcelium

bedros, 2021-03-10:
when compiling Verilog files, it's common to include .f file which lists  all files and  includes switches like "-v" , "-y" "+libext"

Please have a support for filelist as serious projects commonly use them and I cannot use cocotb without such a feature.

Regards,
Bedros

ktbarrett, 2021-03-10:
You don't have to use the makefiles if they don't support all the features you need. It's also likely `SIM_ARGS` does exactly what you want.

sjalloq, 2021-03-10:
@bedros , you'll find that the Makefiles are a bit broken and don't let you just specify a .f file for the COMPILE_ARGS.  They require you to specify at least something in VERILOG_SOURCES or else it does something silly like try to compile for VHDL or something.  I forget the exact failure mode.

So we work around it by just specifying the top level testbench file in VERILOG_SOURCES and pass a .f with all the source RTL, or as is the case below, we also have some properties and bind files we pass in.  Something like this:

```makefile
COMPILE_ARGS = -svinputport=net -sv -f ../nvm.f

ifeq ($(ASSERT_ON),1)
COMPILE_ARGS += +libext+.v+.vlib +define+ASSERT_ON +define+OVL_ASSERT_ON -y $(AHBPC_PATH) -y $(OVL_PATH) +incdir+$(OVL_PATH)
endif

SIM_ARGS = -l cocotb.log -no_autoacc -voptargs="-access=rw+/." -voptargs="+acc" -suppress 12130 -debugdb

VERILOG_SOURCES = \
	../properties/digitalsubsystem_nvm_hs_prop.sv \
	../properties/bind_digitalsubsystem_nvm_hs_prop.sv \
	./sv/tb_nvm.sv

default: _all

include $(shell cocotb-config --makefiles)/Makefile.sim

_all: clean ral regs vc all
```

bedros, 2021-03-10:
@sjalloq  thanks for the comment, this would help a lot.

closed 2021-03-10 by bedros