cocotb/cocotb #2462: ipython_support and pytest assertion rewriting don't work toegther

ktbarrett, 2021-03-11:
When running `make MODULE=cocotb.ipython_support` while in any of the examples or tests I get the following error.

```
     0.00ns ERROR    cocotb.regression                         regression.py:408  in _score_test                     Test Failed: run_ipython (result was OSError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/ipython_support.py", line 88, in run_ipython
                                                                                                                         await embed(user_ns=dict(dut=dut))
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/ipython_support.py", line 77, in embed
                                                                                                                         await run_shell()
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 671, in wrapper
                                                                                                                         return waiter.result  # raises if there was an exception
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/scheduler.py", line 104, in result
                                                                                                                         return self._outcome.get()
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/outcomes.py", line 53, in get
                                                                                                                         raise self.error
                                                                                                                       File "/home/kaleb/dev/cocotb/cocotb/ipython_support.py", line 76, in run_shell
                                                                                                                         shell()
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/IPython/terminal/embed.py", line 228, in __call__
                                                                                                                         self.mainloop(local_ns, module, stack_depth=stack_depth,
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/IPython/terminal/embed.py", line 324, in mainloop
                                                                                                                         self.interact()
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/IPython/terminal/interactiveshell.py", line 547, in interact
                                                                                                                         code = self.prompt_for_code()
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/IPython/terminal/interactiveshell.py", line 290, in prompt
                                                                                                                         lines = [input(prompt_text)]
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/IPython/utils/py3compat.py", line 147, in input
                                                                                                                         return builtin_mod.input(prompt)
                                                                                                                       File "/home/kaleb/miniconda3/lib/python3.8/site-packages/_pytest/capture.py", line 217, in read
                                                                                                                         raise OSError(
                                                                                                                     OSError: pytest: reading from stdin while output is captured!  Consider using `-s`.
```

After commenting out the following section I drop into an ipython shell successfully.

https://github.com/cocotb/cocotb/blob/98f9184a5acce44e5c0ebcf40222a15be524b7dd/cocotb/__init__.py#L290-L305

eric-wieser, 2021-03-12:
I don't really understand why output capturing is being enabled here. What version of pytest are you using, and does the assertion rewriting still work properly for you in non-ipython cases?

marlonjames, 2021-03-12:
I'm not seeing the error in WSL2, but I'm not using conda.

Taking the hint from the error, we could use
```py3
pytest_conf = Config.fromdictargs({}, ['-s'])
```
to ensure there is no capture.

PR #2469 (2021-03-18): Configure pytest to not capture output
Closes #2462.

When using pytest for assertion rewriting, configure to not capture output and consequently not affect `stdin`.

By default, pytest captures `stdin` and `stdout`, and changes `stdin`: https://docs.pytest.org/en/stable/capture.html#default-stdout-stderr-stdin-capturing-behaviour

Also fix the test.
closed 2021-03-18 by marlonjames