cocotb/cocotb #2467: cocotb no longer installing 32bit properly (1.5.0)

stezskal, 2021-03-12:
32bit installation working with 1.4.0 but breaks with 1.5.0
linux centos
Python I tried both miniconda 32bit and a conda virtual environment.

In the attached console log you can see at the very beginning 1.4.0 installs OK and 1.5.0 looks like it has some mixed 32-bit and 64-bit libraries or something along those lines.
[cocotb_1_4_0_good_1_5_0_bad.txt](https://github.com/cocotb/cocotb/files/6131776/cocotb_1_4_0_good_1_5_0_bad.txt)

themperek, 2021-03-13:
Did you follow this: https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions ?

Seems to work for me on centos 7 + conda.

ktbarrett, 2021-03-13:
Looks like an environment issue. The `ld` in your miniconda is loading a 64 bit support library from your system.

stezskal, 2021-03-13:
Thanks for feedback.

I haven't seen those instructions, I'll investigate shortly. 

I agree it seems like an env issue.. except that release 1.4.0 installs ok with my env, but not 1.5.0.  
I'll let you know what I find.

ktbarrett, 2021-03-16:
There are *many* changes to the build system between 1.4.0 and 1.5.0. I don't find it that surprising that a seemingly innocuous environment issue caused problems. The library it is incorrectly loading is liblto. Compiling with LTO by default was added in 1.5.0. You might want to check your 32 bit environment's compiler supports LTO.

stezskal, 2021-03-16:
Ok I see.  I probably got somewhat lucky that the existing env and toolchain worked preciously then.  Thanks for pointing out liblto and LTO.. that gives me something to focus on.

stezskal, 2021-04-19:
We haven't had time to dive into this and are using a previous version of cocotb 1.4.0.  We hope to dive into this in upcoming few weeks.

stezskal, 2021-05-26:
We finally had some time to pursue this and following the instructions in https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions seems to provide the proper environment for the latest version.

closed 2021-05-26 by stezskal