cocotb/cocotb #2471: cocotb causing Verilator erro %Error-TIMESCALEMOD

jg-fossh, 2021-03-14:
cocotb 1.5.0 (and rc2)
OS : Ubuntu 20.10
Python : 3.8
uvm-python : 0.0.2

After the recent upgrade to the latest version of cocotb Verilator will always issue error %Error-TIMESCALEMOD. Setting `timescale on all modules did not help, neither passing --timescale or    --timescale-override  from the Makefile.
Looking at the command line it also looks like cocotb is passing it's own timesacle "-lcocotbvpi_verilator" --trace --trace-structs --timescale 1ns/1ps"

This issue did not appear in version 1.4.0, the problem with rolling back to 1.4.0 is that cocotb-bus seems to be different or pointing to a newer version of cocotb than 1.4.0, so it becomes a process of selective installing and reinstalling until one gets the lucky combination.

cmarqu, 2021-03-14:
Which Verilator version is this? Right now, cocotb requires 4.106 exactly, and 4.108 does not work. See https://github.com/cocotb/cocotb/issues/2300.

jg-fossh, 2021-03-14:
Running
$ verilator --version
Verilator 4.106 2020-12-02 rev UNKNOWN_REV

Downloaded from https://github.com/verilator/verilator/releases/tag/v4.106

ktbarrett, 2021-03-14:
Have you confirmed that the timescale argument is what is causing problems by removing it from the cocotb 1.5 makefile? Can you post a minimal design that causes the error? We aren't seeing that issue with our files.

jg-fossh, 2021-03-15:
When you ask to remove the timescale flag from the the cocotb 1.5 makefile, do you mean my simulation makefile? If this is the case I cannot because it seems to be introduce by cocotb 1.5.

This is an example repo : https://github.com/jg-fossh/Goldschmidt_Integer_Divider.git
To run the simulation go into the sim directory and run `make`. You will probably get a warning from verilator, simply run `make` again and it will ignore the design warnings and simulate.
If you are changing back and forth cocotb version 1.4 and 1.5 you will probably need to delete the pycache folders (there are two more pycache folders within the sim/externals/Wishbone_* directories)  since there is no `make clean` in the Makefile.

ktbarrett, 2021-03-15:
What I meant was clone the cocotb source and do an editable install (`pip install -e .`) then remove the bits from the cocotb makefile to test if that changes anything. Also, did you remember to clean the directory between runs? Verilator builds C++ sources that aren't rebuilt by default if the makefile changes.

jg-fossh, 2021-03-16:
I confirm by commenting out line 43 of cocotb/share/makefiles/simulators/Makefile.verilator fixes the issue. The line commented out: 
EXTRA_ARGS += --timescale $(COCOTB_HDL_TIMEUNIT)/$(COCOTB_HDL_TIMEPRECISION)

Thanks.

ktbarrett, 2021-03-16:
@jg-fossh This is likely a Verilator issue, that still exists in the current stable release issue based on https://github.com/verilator/verilator/issues/2763#issuecomment-792495487. Open an issue in the Verilator repo and link to this issue. 

We aren't having an issue with our tests because they all only have one file. Multiple files seem to trigger the issue. My guess is that `--timescale` is setting the timescale in each file as it goes along, and the error mechanism is seeing that the previous file "has a timescale" and throws the error. Not sure why setting a timescale in each file isn't working. Using `--timescale-override` is also not work. The error detection logic must be buggy.

Your best bet for now is to copy `Makefile.verilator` into your project, remove line 43, and in your project Makefile replace `include $(shell cocotb-config --makefiles)/Makefile.sim`  with `include ./Makefile.verilator`.

wsnyder, 2021-03-17:
Verilator has this fixed (I think) please validate it in context of cocotb and if passes presumably can close, thanks.

ktbarrett, 2021-03-18:
Confirmed it can get passed that error. Looks like a lot of internals changed? I'm getting compile errors on master.

jg-fossh, 2021-03-18:
Yes Verilator has fixed the issue caused by the --timescale flag in version 4.201. 
Since cocotb depends on Verilator 4.106 ( in which the issue still persists ) I don't know what is the guideline for closing the issue. 
We do have a fix documented in this thread, if that enough any cocotb member or contributor  please ratify or close this.

ktbarrett, 2021-03-18:
We have a workaround for using v4.106 and have fixed the issue upstream. I think it's resolved.

closed 2021-03-18 by ktbarrett