cocotb/cocotb #2473: Adding BusDriver makes writes fail for verilator

tpambor, 2021-03-15:
cocotb: `83476d8b`
cocotb-bus: `1c6b2ac2`
verilator: `v4.106`

With `BusDriver`:
```
class FooDriver(BusDriver):
    _signals = []
    _optional_signals = []

    def __init__(self, entity, name, clock, **kwargs):
        BusDriver.__init__(self, entity, name, clock, **kwargs)

@cocotb.test()
async def bad(dut):
    dut_in = FooDriver(dut, "s_axis", dut.aclk)

    dut.m_axis_tready <= 1
    await Timer(10, units="ns")
    assert dut.m_axis_tready.value == 1
```

```
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:99   in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:247  in _embed_sim_init                 Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:247  in _initialise_testbench_          Running on Verilator version 4.106 2020-12-02
     0.00ns INFO     cocotb                                      __init__.py:253  in _initialise_testbench_          Running tests with cocotb v1.6.0.dev0 from /usr/local/lib/python3.9/dist-packages/cocotb
     0.00ns INFO     cocotb                                      __init__.py:274  in _initialise_testbench_          Seeding Python random module with 1615828538
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:239  in gpi_get_root_handle             Looking for root handle '(null)' over 1 implementations
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:245  in gpi_get_root_handle             Got a Root handle (bug) back from VPI
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:53   in check_and_store                 Checking bug exists
     0.00ns DEBUG    cocotb.bug                                    handle.py:119  in __init__                        Created
     0.00ns WARNING  cocotb                                      __init__.py:293  in _initialise_testbench_          Pytest not found, assertion rewriting will not occur
     0.00ns DEBUG    cocotb.regression                         regression.py:177  in _discover_tests                 Python Path: ,/home/fpga/workspace/tests,/usr/lib/python39.zip,/usr/lib/python3.9,/usr/lib/python3.9/lib-dynload,/usr/local/lib/python3.9/dist-packages,/usr/lib/python3/dist-packages
     0.00ns DEBUG    cocotb.regression                         regression.py:178  in _discover_tests                 PWD: /home/fpga/workspace/work/tests
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test tests.hdl.bug.bad
     0.00ns INFO     cocotb.regression                         regression.py:477  in _start_test                     Running test 1/1: bad
     0.00ns INFO     cocotb.test.bad.0x7f08d6128430            decorators.py:312  in _advance                        Starting test: "bad"
                                                                                                                     Description: None
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:265  in __gpi_get_handle_by_name        Searching for aclk
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:274  in __gpi_get_handle_by_name        Checking if aclk is native through implementation VPI
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:283  in initialise                      VPI: aclk initialized with 1 elements
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:285  in __gpi_get_handle_by_name        Found aclk via VPI
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:53   in check_and_store                 Checking bug.aclk exists
     0.00ns DEBUG    cocotb.aclk                                   handle.py:119  in __init__                        Created
     0.00ns DEBUG    cocotb.bug                                    handle.py:244  in _discover_all                   Discovering all on bug
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:651  in VpiIterator                     vpi_iterate type=36 returned NULL
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:651  in VpiIterator                     vpi_iterate type=114 returned NULL
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:662  in VpiIterator                     Created iterator working from 'bug' with type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:854  in next_handle                     vpi_scan found 'bug.aclk'
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:283  in initialise                      VPI: aclk initialized with 1 elements
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:382  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:53   in check_and_store                 Checking bug.aclk exists
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:60   in check_and_store                 Found duplicate bug.aclk
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:854  in next_handle                     vpi_scan found 'bug.advance'
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:283  in initialise                      VPI: advance initialized with 1 elements
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:382  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:53   in check_and_store                 Checking bug.advance exists
     0.00ns DEBUG    cocotb.advance                                handle.py:119  in __init__                        Created
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:854  in next_handle                     vpi_scan found 'bug.areset'
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:283  in initialise                      VPI: areset initialized with 1 elements
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:382  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:53   in check_and_store                 Checking bug.areset exists
     0.00ns DEBUG    cocotb.areset                                 handle.py:119  in __init__                        Created
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:854  in next_handle                     vpi_scan found 'bug.current_sof'
[... 246 lines cut ...]
    11.00ns INFO     cocotb.regression                         regression.py:373  in _score_test                     Test Passed: good
    11.00ns INFO     cocotb.regression                         regression.py:496  in _log_test_summary               Passed 1 tests (0 skipped)
    11.00ns INFO     cocotb.regression                         regression.py:569  in _log_test_summary               **************************************************************************************
                                                                                                                     ** TEST                          PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
[... cut ...]
     -.--ns DEBUG    cocotb.gpi                         ..mbed/gpi_embed.cpp:187  in _embed_sim_cleanup              Returning to simulator
     -.--ns DEBUG    cocotb.gpi                         ..b/gpi/GpiCbHdl.cpp:82   in run_callback                    Generic run_callback done
- /usr/local/lib/python3.9/dist-packages/cocotb/share/lib/verilator/verilator.cpp:118: Verilog $finish
- /usr/local/lib/python3.9/dist-packages/cocotb/share/lib/verilator/verilator.cpp:118: Second verilog $finish, exiting
```

marlonjames, 2021-03-15:
This is #2244, an issue with Verilator VPI.

The cause is this line in `Bus`:

https://github.com/cocotb/cocotb-bus/blob/1c6b2ac2db567d7e352e9a8cf4216b5fe7029035/src/cocotb_bus/bus.py#L59

`dir()` causes iteration over the handle to discover sub-handles, which hits the bug.

ktbarrett, 2021-04-18:
Not our bug and we are already tracking it.

closed 2021-04-18 by ktbarrett