cocotb/cocotb #2474: Simulator crash while running COCOTB with xcelium

thmani, 2021-03-16:
Hi All,
I am trying to run my first cocotb simulations using xcelium. I am stuck at tool crash issue.

`
[image]

`

imphil, 2021-03-16:
The trace shows that you're using Python 2.7 and something is a 32 bit version. Did you ensure that you are running cocotb with Python 3? How did you install cocotb?

ktbarrett, 2021-03-16:
This is an environment issue, but I can't tell who's at fault here.

Not finding "encodings" usually implies PYTHONHOME is being set to something different from where the python executable/libpython is found. I also see in the *internal* traceback, Python. It, is also screwed up using two different Python installations. I'm wonder if Cadence is using Python internally and is setting PYTHONHOME just like they set LD_LIBRARY_PATH? Pinging @cmarqu can you confirm?

But it seems like your Xcelium install is also broken since its using multiple Python installations, including a system installation of Python 2.7 that's 32 bit? What OS are you using? "Linux" is a kernel. Red Hat 6? Ubuntu 16.04? Etc?

cmarqu, 2021-03-16:
No, my Xcelium does not touch  my `PYTHONHOME` which is set to the Anaconda location that I'm using.
FWIW, I checked this with `make SIM=xcelium SIM_ARGS="-input \"@env | grep PYTHON\" -input \"@exit\""` in the `simple_dff` example.

thmani, 2021-03-17:
Hi All,
Similar issue was reported by @juliusbaxter
https://github.com/cocotb/cocotb/issues/1945 

**Workround proposal was:
It indeed did turn out to be an RPATH problem in xrun and xmelab. They bundle a Python3 install with the Xcelium install, the RPATH is set to their local lib64 area, and when running with LD_DEBUG=libs it was clear it was always picking up libpython3.6m.so.1.0 from its local install and not the system one.

I've managed to make cocotb 1.4 work against Xcelium 1909002 by moving those Python 3 libraries out of the Xcelium local lib64 directory.** 

I am trying to move Python libraries out of xcelium library. Please help me if anyone knows how to do it.
I am new to python world :)

themperek, 2021-03-17:
>I've managed to make cocotb 1.4 work against Xcelium 1909002 by moving those Python 3 libraries out of the Xcelium local lib64 directory._

It should not be needed to do anything like this for 1.5. But I am not 100% sure.

thmani, 2021-03-18:
@ktbarrett 
Which python3.6 --> ./cocotb/first_ctb/bin/python3
cocotb-config --libpython --> /usr/lib64/libpython3.6m.so.1.0

@themperek 
I installed cocotb using "
[image]

I think in 1.5.0 version they fixed libpython issue using embed.cpp code(not sure about this)
But I dont see this file in my cocotb installation. I only see the verilator
first_ctb> ls lib/python3.6/site-packages/cocotb/share/lib/
verilator/

thmani, 2021-03-18:
Also I do not see below files in my cocotb library. Could this be a problem as I see from error message that crash happens while executing coctotb.gpi method?

[image]

ktbarrett, 2021-03-18:
@thmani You won't see any of those files in your installation, they are compiled into shared libraries under `$(cocotb-config --prefix)/cocotb/libs`. The `Verilator.cpp` is necessary though for Verilator support.

`cocotb-config --libpython` seems to be correctly finding libpython3.6m.so in your environment. So the correct libpython is being loaded and the correct Python libraries (your system ones) should be used. ***Unless Xcelium beats you to the punch***. I think you are going to need to do some library level debugging (Adding `LD_DEBUG=libs` environment variable) to see if Xcelium is loading libpython before us. @cmarqu Could you help us again?

ktbarrett, 2021-03-18:
@thmani Could you also send me a dump of you running `LD_DEBUG=libs xrun`?

thmani, 2021-03-18:
@ktbarrett  sent you the log file after setting environment variable LD_DEBUG=libs

ktbarrett, 2021-03-20:
I see it "searching" for `libpython3.6m.so` as a dependency of `libcocotbvpi_ius.so` before the VPI library is initialized. This is not cocotb 1.5 behavior. Ensure you have cocotb 1.5.0 install in the environment where you think it is.

thmani, 2021-03-23:
@ktbarrett 
Now I have installed cocotb 1.5.1, but I still get the same error. any more ideas to try out?

themperek, 2021-03-23:
@thmani Please minimum provide log file.

thmani, 2021-03-23:
@themperek : Please let me know how I can share the log file directly to you. My email:thmani@gmail.com

themperek, 2021-03-23:
It looks like something nontrivial environment miss-configuration. 
I would suggest installing and using [conda](https://docs.conda.io/en/latest/miniconda.html) (in local/home folder) that should isolate you from the system.

thmani, 2021-03-24:
@themperek, Sorry for the very basic question..
I tried installing cocotb using conda (conda install cocotb). But cocotb is not found in default channels.
I have downloaded the cocotb 1.5.1 code. Any idea how to install this using conda?

ktbarrett, 2021-03-24:
@thmani `pip install cocotb`

thmani, 2021-03-25:
@ktbarrett. Thanks for the quick feedback. I started all over again without virtual env this time.
I tried below steps:
1. mkdir cocotb
2. pip3.6 install cocotb --user
3. setenv PATH ${PATH}:/$USER/.local/bin
4. cocotb> cocotb-config --version
Traceback (most recent call last):
  File "/$USER/.local/bin/cocotb-config", line 7, in <module>
    from cocotb.config import main
  File "/$USER/.local/lib/python3.6/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/$USER/.local/lib/python3.6/site-packages/cocotb/handle.py", line 38, in <module>
    from cocotb import simulator
**ImportError: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /$USER/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so)**

5. strings /lib64/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
XXXXXXXXX ....
GLIBCXX_3.4.19
GLIBCXX_DEBUG_MESSAGE_LENGTH

6. install gcc/4.9.1
7. cocotb-config --version
Traceback (most recent call last):
  File "/$USER/.local/bin/cocotb-config", line 7, in <module>
    from cocotb.config import main
  File "/$USER/.local/lib/python3.6/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/$USER/.local/lib/python3.6/site-packages/cocotb/handle.py", line 38, in <module>
    from cocotb import simulator
**ImportError: /opt/dsard/gcc/4.9.1/lib64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /$USER/.local/lib/python3.6/site-packages/cocotb/libs/libgpi.so)**
8. strings /opt/dsard/gcc/4.9.1/lib64/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
XXXXXXXXX ....
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_DEBUG_MESSAGE_LENGTH

any idea why this doesnt work even after updating gcc version?

thmani, 2021-03-25:
Hi All, Thanks a lot for your support. I fixed the issue after moving out of local virtual env and swithching to  gcc version 8.2.0.
Thanks to @themperek and @ktbarrett !

closed 2021-03-25 by thmani

ktbarrett, 2021-03-25:
@thmani That's good news! We will investigate why virtualenv might be causing problems.

Also, your issue with GLIBCXX_3.4.21 sounds like an environment issue as well. Your cocotb was compiled with a newer GCC and against a newer libstdc++ than the one it is picking up from your environment. You should make sure the same GCC/Clang/MSVC you compile with is the one in your environment when executing.