cocotb/cocotb #2478: Weird ClockCycles behaviour

aignacio, 2021-03-17:
Hey everyone,

I'm having an unexpected behaviour during execution of a test using ClockCycles to do the reset in my dut. My design+tb has two clocks, one that's faster than the other and I always use the slowest one to hold the reset of both segments of the design for 2 cc. Although I use the correct signal to wait for the slowest clock, it seems that the simulation is using always the same clock (in this case noc_clk) as reference to wait inside the ClockCycles. Please check the snippet below:

Based on the option used in the regression, I invert the clocks that I'm using in the dut.
https://github.com/aignacio/ravenoc/blob/master/tb/common_noc/testbench.py#L123-L130
```python
    async def setup_clks(self, clk_mode="NoC_slwT_AXI"):
        self.log.info(f"[Setup] Configuring the clocks: {clk_mode}")
        if clk_mode == "NoC_slwT_AXI":
            cocotb.fork(Clock(self.dut.clk_noc, *noc_const.CLK_100MHz).start())
            cocotb.fork(Clock(self.dut.clk_axi, *noc_const.CLK_200MHz).start())
        else:
            cocotb.fork(Clock(self.dut.clk_axi, *noc_const.CLK_100MHz).start())
            cocotb.fork(Clock(self.dut.clk_noc, *noc_const.CLK_200MHz).start())
```

..and the reset method does the same, checking what's the slowest one to hold for 2cc
https://github.com/aignacio/ravenoc/blob/465a5eb2bf0c242c0af5f9f4c1e3169324d89f51/tb/common_noc/testbench.py#L139-L149
```python
    async def arst(self, clk_mode="NoC_slwT_AXI"):
        self.log.info("[Setup] Reset DUT")
        self.dut.arst_axi.setimmediatevalue(0)
        self.dut.arst_noc.setimmediatevalue(0)
        self.dut.axi_sel.setimmediatevalue(0)
        self.dut.arst_axi <= 1
        self.dut.arst_noc <= 1
        if clk_mode == "NoC_slwT_AXI":
            await ClockCycles(self.dut.clk_noc, noc_const.RST_CYCLES)
        else:
            await ClockCycles(self.dut.clk_axi, noc_const.RST_CYCLES)
```

here's what I got when I run the regression
[image]
In this image it's pretty clear it's always using clk_noc as reference on the trigger. (ClockCycles(self.dut.clk_noc, noc_const.RST_CYCLES))

From my understanding the simulation should always take the same amount of time for the reset in both cases, right?

Cocotb - v1.5
Verilator - 4.106

ktbarrett, 2021-03-17:
I think the choice of the NoC clock being exactly 2x the AXI clock is making it seem like the issue is in one place, when it's really in another.

In the first reset, the first clock edge seen by `ClockCycles` is at time = 0, and the second on the next clock edge. To achieve what I think you are intending you should first synchronize with the clock before doing `ClockCycles`.

aignacio, 2021-03-17:
Thanks @ktbarrett,
guess you were right, I changed to three cc and it makes sense your answer. To fix as you suggested you recommend me to wait a rising edge of the slower clock before launching the resets?
[image]

aignacio, 2021-03-17:
just adding one more question if you don't mind =). In the statement below, it should wait till an irq gets triggered before it does an AXI read operation but I'm seeing that in the sims, it's passing through the if+trigger whatsoever is the value there...

https://github.com/aignacio/ravenoc/blob/master/tb/test_ravenoc_basic.py#L46-L47
```python
    if int(tb.dut.irqs_out) == 0:
        await with_timeout(Edge(tb.dut.irqs_out), *noc_const.TIMEOUT_IRQ)
```
here's the waves in the case that fails in the test
[image]
as you can see the read starts before the irqs_out switches from 0 to something, is the snippet below wrongly implemented the intended behaviour?

ktbarrett, 2021-03-17:
The snippet looks good, that should do what you intend. What's the value of `int(tb.dut.irqs_out)` as seen by cocotb? Could you add a `print(int(tb.dut.irqs_out))` before the if statement?

aignacio, 2021-03-17:
I've added in the log st the print of the value when it fails, apparently it's showing as zero...

https://github.com/aignacio/ravenoc/blob/724f140ece75b6ef501997ce1ac210144a02d8c4/tb/test_ravenoc_basic.py#L48

```bash
   290.00ns INFO     sim_NoC_slwT_AXI_coffee            ..t_ravenoc_basic.py:49   in run_test                        Value IRQS before read 000000000000
```

aignacio, 2021-03-17:
ps: sometimes the test passes sometimes not
ps.2: on verilator 4.110 the simulation hangs indefinitely, so I'm still using 4.106

ktbarrett, 2021-03-17:
For now, only 4.106 "works". There are a couple bugs in Verilator that need to be fixed.

If the value is 0, and the if statement is followed, and the waiting for edge occurs, what's the problem? Are you saying that `Edge` is returning immediately?

aignacio, 2021-03-17:
Yes, the edge returns immediately with the value still being zero, the log above is after the await with_timeout(Edge....

ktbarrett, 2021-03-17:
I could track that down, but odds are it's a Verilator bug. Have you tried using the code with another simulator?

aignacio, 2021-03-17:
Thanks for the answer @ktbarrett,
no problem, I'm still willing to move on with Verilator once it's the only one SV open source supported  for now. I couldn't try another simulator (I wish to use xcelium but I'm out of licenses for now) and for the moment I'm gating the axi read by doing something like this:
```python
 while int(tb.dut.irqs_out) == 0:
        await RisingEdge(tb.dut.clk_noc)
```
Though I still need to embedded the time to avoid any hangs. My guess is that there's an issue with the _EdgeBase class > GPI with arrays in Verilator. Who's keeping track of Verilator support in Cocotb for now? Just for heads up about this unexpected behaviour

ktbarrett, 2021-03-18:
There is no one dedicated to improving cocotb support in Verilator, it's volunteer effort like the whole cocotb project. You can see the current issues with Verilator by filtering cocotb's issues page with the `category:simulators:verilator` label.

I'm fairly certain that `Edge` triggers on 1D logic arrays should work, but you can try the following which creates an Edge trigger on each bit individual and wait until any of them fire.
```python
await First(*(Edge(bit) for bit in tb.dut.irqs_out))
```

aignacio, 2021-03-18:
tks for the idea, I tried to implement like you suggested plus adding the timeout wrapper and I'm seeing timeout exceptions (raise cocotb.result.SimTimeoutError):
```python
    await with_timeout(First(*(Edge(bit) for bit in tb.dut.irqs_out)), *noc_const.TIMEOUT_IRQ)
```
Isn't possible to encapsulate like this?

> My current ugly solution for now is this:
https://github.com/aignacio/ravenoc/blob/14932c7e1b1173e11f864e5bdc639af1ffdbfc9e/tb/test_ravenoc_basic.py#L55-L64

ktbarrett, 2021-03-18:
I'm guessing those timeouts aren't expected? I don't know what to tell you... If your current solution works, stick to it. Things will probably get better as time goes on.

aignacio, 2021-03-18:
yeap, timeouts are not expected..np @ktbarrett, for the moment I'm going to proceed with this workaround, anyway tks a lot for the support! Just before closing the issue, from your understanding arrays on top are still not supported on the latest Verilator? I'm asking cause I needed to encapsulate my dut in a muxed wrapper to talk with different axi channels i/fs

ktbarrett, 2024-04-07:
Try with the newest versions of verilator and cocotb. Reopen if the problem persists.

closed 2024-04-07 by ktbarrett