cocotb/cocotb #2480: Binary representation with Cocotb 1.5

Botnic, 2021-03-18:
In the class BinaryValue there is the possibility to pass during initialization a parameter called binaryRepresentation
` def __init__(self, value=None, n_bits=None, bigEndian=True,
                 binaryRepresentation=BinaryRepresentation.UNSIGNED,
                 bits=None):`

This setting can later be read out over the 'binaryRepresentation' variable.
While this was editable in cocotb1.4, changing this variable in 1.5 has no influence on the conversion functions any more. 
The representation can only be set during creation of the object.

ktbarrett, 2021-03-18:
This was anticipated, from https://github.com/cocotb/cocotb/pull/1286#discussion_r364012422. I am surprised it worked for you at any point.

> I would suggest to
> 
> * make the variable binaryRepresentation private, so users don't try to change it any more.
>   or
> * update the conversion function when binaryRepresentation is chagned

For 1.5.x we should maintain backwards compatibility, so the second is the better option. Could you introduce a PR for 1.5.1 to take us back to the old behavior?

As for 2.0. We have stated that `BinaryValue` will be tossed and replaced with a much more usable set of types (#2059). The representation *will not* be associated with the object. They really isn't a good reason to. Instead, we will support multiple conversion functions:

```python
vec = LogicArray[1:'to':8].from_twos_complement(-123)
assert vec.to_unsigned() == 133
```

Or something similar to that. We haven't exactly nailed down how arrays are going to look. Or if we should have a single `to_int`, `from_int` pair that take representation strings or enums.

Botnic, 2021-03-19:
Thank you for fixing this issue so fast!

PR #2481 (2021-03-19): Revert BinaryValue change
Closes #2480. Reintroduces to the 1.5 release the ability of the user to change the `binaryRepresentation` of a `BinaryValue` object after its creation.
closed 2021-03-19 by Botnic