cocotb/cocotb #2490: Deprecate __setattr__ overload for setting a handle value

ktbarrett, 2021-03-26:
We already have two other ways to accomplish this and the [Zen of Python](https://www.python.org/dev/peps/pep-0020/) states "There should be one-- and preferably only one --obvious way to do it."

```python
handle.value = value
handle <= value
```

The `__setattr__` syntax depends on there being at least one additional level of hierarchy to be able to set a signal; and this can result in unintended behavior with newbies to Python. I have seen this happen at least 3 times.

```python
dut.handle = value  # okay
handle = value  # doh!
```
 
I think this should behavior should be deprecated.

imphil, 2021-03-26:
I agree with having only one way of doing things, and that the current behavior is very error-prone. (handle = value is something I've seen often as well).

> I think this should behavior should be deprecated.

Which of the two behaviors?

eric-wieser, 2021-03-26:
The third behavior, which @ktbarrett didn't give an example of but otherwise described

closed 2021-04-02 by ktbarrett

mnemocron, 2023-07-07:
Ok, but then what is the correct way to do it anyway?

If I have this:
```python
@cocotb.test()
async def stim_input(dut):
	cocotb.start_soon(Clock(dut.clk, ACLK_PERIOD, units="ns").start())
	dut.i_resetn.value = 1
```
It will throw the warning
```
test_reset_deasert_sync.py:18: DeprecationWarning: Setting values on handles using the ``dut.handle = value`` syntax is deprecated. Instead use the ``handle.value = value`` syntax
```

but then deleting `dut` just throws an error
```
Traceback (most recent call last):
   File "/test_reset_deasert_sync.py", line 14, in stim_input
   i_resetn.value = 1
   UnboundLocalError: local variable 'i_resetn' referenced before assignment
```

ktbarrett, 2023-07-08:
@mnemocron I have no idea why the code you posted would cause the warning. You are not using __setattr__ syntax. Are you sure the file and line number in the warning correspond to the code you posted?

mnemocron, 2023-07-11:
Yes, the line number corresponds to the line `dut.i_resetn.value = 1`.

I attached the basic design i created, where it occurs.
[reset_deasert_sync.zip](https://github.com/cocotb/cocotb/files/12013049/reset_deasert_sync.zip)

```
Ubuntu 20.04.6 LTS focal on Windows 10 x86_64
cocotb-config --version
1.8.0
```

mnemocron, 2023-07-15:
@ktbarrett yes, that is correct. Thanks :)

PR #2493 (2021-04-02): Deprecate setting handle values with __setattr__ syntax
Closes #2490. Adds deprecation for `dut.handle = value` syntax.