cocotb/cocotb #2495: Using asyncio Queues

raysalemi, 2021-03-28:
Hi, (@ktbarrett )

I'm looking to use the new aysncio queues in an example and I'm not certain how to import them. It appears that cocotb queue.py is intended to override the normal queue.py.  Is this true?  I tried using cocotb.Queue instead of queue.Queue and got an error. But now queue.Queue hangs.

It appears that **cocotb** is making the assumption that the user will want to use nothing but coroutines in a testbench. This isn't a good assumption. For example pyuvm does not use coroutines, but interacts with them through the normal queue.Queue.

I think the aysncio queue should be named AsyncioQueue to avoid confusion with the existing Queue class.

Ray

eric-wieser, 2021-03-28:
Regarding the naming; `cocotb.queue.Queue` is designed to largely match the built-in `asyncio.Queue`.

raysalemi, 2021-03-28:
> Hi, (@ktbarrett )
>
> I'm looking to use the new aysncio queues in an example and I'm not certain how to import them. It appears that cocotb queue.py is intended to override the normal queue.py. Is [...]

Nevermind. I found cocotb.queue.Queue.
RTFM!

closed 2021-03-28 by raysalemi