cocotb/cocotb #2496: Error on line 111 of Makefile trying to set TOPLEVEL_LANG

raysalemi, 2021-03-28:
This Makefile:

```
CWD=$(shell pwd)
COCOTB_REDUCED_LOG_FMT = True
SIM ?= icarus
VERILOG_SOURCES =$(CWD)/hdl/verilog/tinyalu.sv
VHDL_SOURCES = $(CWD)/hdl/vhdl/single_cycle_add_and_xor.vhd	\
$(CWD)/hdl/vhdl/three_cycle_mult.vhd \
$(CWD)/hdl/vhdl/tinyalu.vhd
MODULE := tinyalu_cocotb
TOPLEVEL := tinyalu
# TOPLEVEL_LANG := verilog
COCOTB_HDL_TIMEUNIT=1us
COCOTB_HDL_TIMEPRECISION=1us
```

Gives this error

```
/home/rsalemi/.local/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:111: *** invalid syntax in conditional.  
```

Meaning this code:

```
ifeq ($(TOPLEVEL_LANG),)

ifneq ($(and $(VHDL_SOURCES),$(if $(VERILOG_SOURCES),,1),)
	TOPLEVEL_LANG := vhdl
else ifneq ($(and $(VERILOG_SOURCES),$(if $(VHDL_SOURCES),,1),)
	TOPLEVEL_LANG := verilog
endif

endif
```

I do note see the error on Macos only Centos:

```
orw-salemi-vm:~/repos/blog_examples/cocotb>  cat /etc/redhat-release
CentOS Linux release 7.6.1810 (Core)
orw-salemi-vm:~/repos/blog_examples/cocotb>
```

```
orw-salemi-vm:~/repos/blog_examples/cocotb>  make --version
GNU Make 3.82
Built for x86_64-redhat-linux-gnu
Copyright (C) 2010  Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
orw-salemi-vm:~/repos/blog_examples/cocotb>
```

ktbarrett, 2021-03-28:
I'm guessing this has to do with the use of the `and` function, which I don't believe was added until gmake 4.0. We should rewrite that bit, RHEL/CentOS 7 is probably the oldest platform we support.

cmarqu, 2021-04-27:
Another user reported the same problem with GNU make 4.1 (https://gitter.im/cocotb/Lobby?at=6087a859be3536391591e893).
The `and` conditional is supported since 3.81 according to https://fossies.org/linux/make/NEWS (line 567) so something else in that original expression seems to be an erroneous corner case that your fix has circumvented.

huang-jl, 2021-04-27:
The problem is because the brackets do not match in `Makefile.inc:111`:
```make
ifneq ($(and $(VHDL_SOURCES),$(if $(VERILOG_SOURCES),,1),)
	TOPLEVEL_LANG := vhdl
else ifneq ($(and $(VERILOG_SOURCES),$(if $(VHDL_SOURCES),,1),)
	TOPLEVEL_LANG := verilog
endif
```

It should look like this instead

```make
ifneq ($(and $(VHDL_SOURCES),$(if $(VERILOG_SOURCES),,1)),)
	TOPLEVEL_LANG := vhdl
else ifneq ($(and $(VERILOG_SOURCES),$(if $(VHDL_SOURCES),,1)),)
	TOPLEVEL_LANG := verilog
endif
```

ktbarrett, 2021-04-27:
@cmarqu RHEL7 is on GNU Make 3.80, so `and` would also not work there. I see the issue with the unbalanced parens, I'm guessing Make introduced a "know what you meant" feature (or maybe just a bug :wink:) that allows the faulty expression to work in a more recent version.

imphil, 2021-05-04:
This bug should be fixed on master and fixed in 1.5.2.

(General note: closing the bug as soon as a problem is fixed on master is the right approach. If we want a backport to a release branch, create a separate issue for the backport, please.)

PR #2515 (2021-04-11): Ensure makefiles work in gmake 3.80 (RHEL7)
Closes #2496. Will cherry-pick onto 1.5 branch after merge. @raysalemi Does this work for you? I'm pretty sure only `and` should have been a problem.
closed 2021-05-04 by imphil