cocotb/cocotb #2497: Fatal crash in Questa 2020.2_1

raysalemi, 2021-03-28:
```
** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
Exit codes are defined in the "Error and Warning Messages"
appendix of the QuestaSim User's Manual.
# Loading sv_std.std
# Loading work.tinyalu(fast)
# Loading work.single_cycle(fast)
# Loading work.three_cycle(fast)
# Loading /home/rsalemi/.local/lib/python3.7/site-packages/cocotb/libs/libcocotbfli_modelsim.so
#      -.--ns INFO     cocotb.gpi                         ..ib/fli/FliImpl.cpp:1071 in cocotb_init                     cocotb_init called
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:77   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       FLI registered
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:240  in _embed_sim_init                 Python interpreter initialized and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:220  in _initialise_testbench_          Running on ModelSim for Questa-64 version 2020.2_1 2020.05
#      0.00ns INFO     cocotb                                      __init__.py:227  in _initialise_testbench_          Running tests with cocotb v1.5.1 from /home/rsalemi/.local/lib/python3.7/site-packages/cocotb
#      0.00ns INFO     cocotb                                      __init__.py:247  in _initialise_testbench_          Seeding Python random module with 1616951681
#      0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test tinyalu_cocotb.test_alu
#      0.00ns INFO     cocotb.regression                         regression.py:472  in _start_test                     Running test 1/1: test_alu
#      0.00ns INFO     cocotb.test.test_alu.0x7fffe9e78e90       decorators.py:313  in _advance                        Starting test: "test_alu"
#                                                                                                                      Description: 

None
#      0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:349  in _embed_sim_event                FATAL: We are calling up again
# End time: 10:14:52 on Mar 28,2021, Elapsed time: 0:00:13
# Errors: 0, Warnings: 0
make[1]: *** [results.xml] Error 211
make[1]: Leaving directory `/home/rsalemi/repos/blog_examples/cocotb'
make: *** [sim] Error 2
orw-salemi-vm:~/repos/blog_examples/cocotb>  vsim --version
Tk initialization failed: no display name and no $DISPLAY environment variable.
orw-salemi-vm:~/repos/blog_examples/cocotb>  vsim --version  -c
Reading [pref.tcl](url)
```

Here is the design and the stack trace:
[ray_example.tar.gz](https://github.com/cocotb/cocotb/files/6217722/ray_example.tar.gz)

raysalemi, 2021-03-28:
I am now seeing this crash even when backreving to cocotb 1.4 in a new venv.

raysalemi, 2021-03-28:
I ran successfully with 2021.1_1 with cocotb 1.4.

ktbarrett, 2021-03-28:
The problem shown in the stacktrace is in [this function](https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/fli/FliImpl.cpp#L274-L376). `+0x788` doesn't really help me there; a line number would be nice? You can run gdb with cocotb by doing `make COCOTB_ATTACH=20`. This should print the PID. Then you can switch into another window and run `gdb attach {PID}`. After that standard gdb debugging can be done and we can see line numbers.

The other part of the problem, and why you are seeing `FATAL: we are calling up again` is explained [here](https://github.com/cocotb/cocotb/issues/2178#issuecomment-723386093). Questa re-enters user code when shutting down, which is poor behavior for a scheduler and is not standards compliant.

closed 2023-01-23 by imphil