cocotb/cocotb #2500: Issue with Reading Records in Riviera from Cocotb

jwprice100, 2021-03-30:
I have a record with members named status and st, both std_logic. When I try to read from st, I get status's value. When I read from status, I also get status's value.  I tried this record on both a port and as a signal, same results. For signals named status and st whom do not belong to a record, the behavior is as expected. I've attached an example test that fails

Cocotb Version: 1.5
Operating System: Centos7 64 bit
Simulator: Riviera 2019.04 64 bit
Python Version: 3.7.6 64 bit self 

[record_test.zip](https://github.com/cocotb/cocotb/files/6232483/record_test.zip)

jwprice100, 2021-03-30:
In my example, renaming st to start results in expected behavior.

marlonjames, 2021-03-30:
I confirmed the test fails on Active-HDL as well, which I expected, as the VSimSA internals are very similar in both simulators.

Did some digging, and Riviera/Active-HDL are returning the same `vhpiHandleT` for `st` and `status` selected names.
So it makes sense that cocotb is getting the same value for both.

My guess is that they are matching names incorrectly, where `st` matches the start of `status`.
cocotb has a similar issue with generate blocks (#2255).

imphil, 2021-03-30:
Are one of you able to submit a bug report to Aldec?

jwprice100, 2021-03-30:
Does this look like an issue with Riviera? If so I can definitely submit a bug report.

marlonjames, 2021-03-30:
Yeah it's a problem with Riviera. We don't have an active support contract with Aldec right now at work, so i'll defer to @jwprice100.

If they ask for a testcase I can put together a pure-VHPI test for them.

jwprice100, 2021-03-30:
Sounds good, I'll file a report in the morning and comment here when I hear back (usually takes a few days in my experience).

imphil, 2021-03-30:
thanks @jwprice100 

(Just FYI: Cocotb has an active support contract with Aldec, let me know if there's something I can forward.)

jwprice100, 2021-04-06:
Quick update. Aldec inquired for more details today and I linked them to this issue. I believe they've moved the issue to the actual development team now which in my experience will take a while to hear back.

jwprice100, 2021-04-11:
Ok this issue is officially fixed in Riviera Pro 2020.10!

closed 2021-04-11 by jwprice100