cocotb/cocotb #2505: How to handle different vhdl libraries

mkzo7, 2021-03-31:
Hi,
I have a question:
Is it possible to run cocotb simulation with specific library (other than work)?
Cheers,
Marcin

themperek, 2021-04-01:
This https://docs.cocotb.org/en/stable/building.html#var-VHDL_SOURCES_%3Clib%3E ?

github-4o, 2021-04-01:
no vhdl libs with modelsim yet, right?

themperek, 2021-04-01:
It was added (#2465) and should work in the development/master branch.

mkzo7, 2021-04-02:
Hi,
We are using Aldec Riviera PRO, will it also work with this simulator?

Cheers,

[Baraja Pty Ltd]

Marcin Kozik
Software Architect | Baraja<http://www.baraja.com/>
Building 91, 14 Julius Ave, North Ryde, NSW, 2113, Australia
D: +61 2 8467 4151

This email and any attachment is confidential between Baraja and the addressee and may be subject to legal professional privilege.
If it has been sent to you in error, please delete it and notify us. Personal information contained in email to us is subject to our Privacy Policy, for more information see Baraja Privacy Policy<http://www.baraja.com/privacy>

themperek, 2021-04-02:
>We are using Aldec Riviera PRO, will it also work with this simulator?

No. One would need to do similar to #2465 in `Makefile.riviera`

closed 2021-05-10 by github-actions[bot]