cocotb/cocotb #2511: Segmentation fault running with Ghdl

gilbertwaltoon, 2021-04-05:
I'm a newbie with cocotb:
I'm running on Windows 64 with Anaconda Python 3.7 and ghdl.
Following the install instructions I've done a
`conda install -c msys2 m2-base m2-make`
and
`pip install cocotb.`
(I don't know if relevant, but the `pip install` failed at first complaining the install required MS Visual C++ Build Tools. Once I'd installed these, the pip install succeeded).

I'm now trying to run the basic adder.vhdl example from the cocotb documentation
In a conda shell I'm doing 
`(py37) C:\Users\DD\Documents\adder\tests>make SIM=ghdl TOPLEVEL_LANG=vhdl`

but getting a Segmentation Error :

`make results.xml`
`make[1]: Entering directory '/c/Users/DD/Documents/adder/tests'`
`\`
`/c/Users/DD/Documents/fpga-toolchain/bin/ghdl -i    --workdir=sim_build --work=work /c/Users/DD/Documents/adder/tests/../hdl/adder.vhdl && \`
`/c/Users/DD/Documents/fpga-toolchain/bin/ghdl -m    --workdir=sim_build -Psim_build --work=work adder
MODULE=test_adder TESTCASE= TOPLEVEL=adder TOPLEVEL_LANG=vhdl \`
`/c/Users/DD/Documents/fpga-toolchain/bin/ghdl -r   --workdir=sim_build -Psim_build --work=work adder --vpi=C:/users/dd/anaconda3/envs/py37/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll`
`loading VPI module 'C:/users/dd/anaconda3/envs/py37/lib/site-packages/cocotb/libs/cocotbvpi_ghdl.dll'
     -.--ns INFO     cocotb.gpi                         ..mbed\gpi_embed.cpp:77   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ..\gpi\GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
VPI module loaded!`
`/c/users/dd/anaconda3/envs/py37/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl`
`make[1]: *** [results.xml] Segmentation fault`
`make[1]: Leaving directory '/c/Users/DD/Documents/adder/tests'`
`c:\users\dd\anaconda3\envs\py37\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed`
`make: *** [sim] Error 2`

themperek, 2021-04-05:
In the past GHDL had issues on windows related to VPI. No idea if it is fixed.

gilbertwaltoon, 2021-04-05:
Hi - The console is reporting 
` VPI registered VPI module loaded`
so that part at least seems maybe o.k.

ktbarrett, 2021-04-05:
Just because it loads the VPI, doesn't mean any later usage could be causing a failure. You can try debugging it? No idea how to go about that on Windows. Does gdb work?

gilbertwaltoon, 2021-04-05:
Hi - 

I managed to get some debug info. out of Microsoft Visual Studio debugger with COCOTB_ATTACH 

As you suggested, it seems ghdl may be the problem (?). 

The debugger reports:

`'Exception thrown at 0x0000000000CA6618 in ghdl.exe: 0xC0000005: Access violation executing location 0x0000000000CA6618.'`

The full report was:

```
Exception thrown at 0x0000000000CA6618 in ghdl.exe: 0xC0000005: Access violation executing location 0x0000000000CA6618.

'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Documents\fpga-toolchain\bin\ghdl.exe'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\dbghelp.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\cocotbvpi_ghdl.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\gpi.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\gpilog.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Documents\fpga-toolchain\lib\libghdlvpi.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\msvcp140.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140_1.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\vcruntime140.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\cocotbutils.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\embed.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\python37.dll'. Symbols loaded.
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\cocotb.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Users\DD\Anaconda3\envs\py37\Lib\site-packages\cocotb\libs\pygpilog.dll'. Module was built without symbols.
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\cryptsp.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\rsaenh.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\bcrypt.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\cryptbase.dll'. 
'ghdl.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. 
The thread 0x3078 has exited with code 0 (0x0).
Exception thrown at 0x0000000000CA6618 in ghdl.exe: 0xC0000005: Access violation executing location 0x0000000000CA6618.
```

I guess there's not much  the Cocotb team can do, and I should report this to ghdl guys?

ktbarrett, 2021-04-05:
Yeah you should. They are going to want a raw VPI test though. 
GHDL's VPI is still very experimental; there are bugs and unimplemented features. You are probably better off using a different simulator for now, if you can.

gilbertwaltoon, 2021-04-17:
Just an update: 
Whilst I haven't found a solution for ghdl in 'pure' Windows, by installing MSYS cocotb and the MSYS Ghdl package from here : https://packages.msys2.org/package/mingw-w64-x86_64-ghdl 
I am now up and running.
So at at least provides a solution for us VHDL/Windows-PC users.

ktbarrett, 2021-04-17:
Installing cocotb into MSYS worked for you? I tried a couple months ago and had issues with their "atypical" Python installation. I guess I should try again.

WSL is also an option if you were willing to go to a Linux-like environment. Many people who want to use Windows aren't willing, which is why I didn't mention it.

ktbarrett, 2021-04-17:
You found a workaround and the original problem is not with us.

closed 2021-04-17 by ktbarrett