cocotb/cocotb #2513: Xunit results.xml files no longer contain error messages from failed tests in cocotb 1.4+

TC01, 2021-04-08:
In cocotb 1.3, and older releases, the xunit results.xml files created by cocotb would include the failure message reporting why a test failed. The output for a failing test in cocotb 1.3 used to look something like this:

```
    <testcase classname="Testbench" name="test_init_regression" ratio_time="10706.29590292502" sim_time_ns="32018.750001" time="2.9906468391418457">
      <failure message="Test failed with random_seed=1617902333" stderr="" stdout="'Failure message'" />
    </testcase>
```

Now, it seems to look like this, in cocotb 1.5:

```
    <testcase classname="Testbench" file="...filename..." lineno="52" name="test_init_regression" ratio_time="11971.774304490144" sim_time_ns="32018.750001" time="2.6745200157165527">
      <failure />
    </testcase>
```

Having the failure messages (the stderr and stdout) made it possible to parse results.xml to find out (approximately) why a test had failed. I'd written a script to do exactly that some time ago; I just tried to dust it off and get it working with a newer cocotb release and discovered this.

I looked through the commit history to find out exactly why this changed. It seems like [this commit](https://github.com/cocotb/cocotb/commit/f8179c23579e9bc034a83bab5d0180453ba8456b) removed `regression._add_failure` in favor of just calling `xunit.add_failure`.  But it was `regression._add_failure` which seems to have been responsible for providing the failure message. It's not clear to me from that commit if this was deliberate or not, and I didn't see anything in the release notes.

I might have been the only person using this feature-- but I thought it was pretty useful. Is there any chance we could re-add the failure messages? Or is it no longer possible as a consequence of the broader scheduler/regression rewrite (which this commit seems to have been part of)?

ktbarrett, 2021-04-08:
We just realized that https://github.com/cocotb/cocotb/pull/2507.

You are the first person to mention it, but I think we probably should add it back in (using a cleaner approach) now that there is some attempt to standardize test reporting on JUnit style reporting and (I think) they expect error messages in the report [OSVR](https://umarcor.github.io/osvb/apis/logging.html#tracking-requirements). @umarcor

eric-wieser, 2021-04-08:
I removed it in [2a89307](https://github.com/cocotb/cocotb/commit/2a893070a901baf41a3315fff2c79ddb0c69f8f1) (#1025), I think by accident.

TC01, 2021-04-08:
Aha! I looked around briefly but didn't see that merge request. Sorry for not complaining sooner. :)

eric-wieser, 2021-04-08:
Note that the behavior before was only to record logs from one specific logger (`cocotb.log`?) to the xml file, as otherwise it ended up far too big.

ktbarrett, 2021-04-08:
I actually think that recording things from `cocotb` logger namespace is a bad idea (maybe besides errors?). Users should be using their own namespace. We need some mechanism to allow users to decide what ends up in the results.xml. I don't think reversion is a good idea, I think we should reimagine this as a new feature.

themperek, 2021-04-08:
Should we patch this for 1.5?