cocotb/cocotb #2516: OSError: reading from stdin while output is captured

toddstrader, 2021-04-12:
When trying to use `pdb` from within a cocotb test I get the above error.  This is when using Verilator with our own verilated runner C++ (i.e. not `verilator.cpp`) and I have so far not been able to reproduce it with cocotb examples.

However, I have tracked the problem down to these two lines in `__init__.py`:
```
            pytest_conf = Config.fromdictargs([], {})
            install_importhook(pytest_conf)
```

If I comment those lines out, I can use `pdb` again.

I'm not familiar with `pytest`.  Is this code supposed to be used all the time or is it only supposed to be executed during the `tests/pytest/` tests?

cmarqu, 2021-04-12:
See #2462.

toddstrader, 2021-04-12:
Nice, that fixed it.  Thanks.

closed 2021-04-12 by toddstrader

ktbarrett, 2021-04-12:
Don't know if you followed the rabbit hole, but the fix was ported back to v1.5.1, so updating should fix the problem.