cocotb/cocotb #2522: Documentation update - python 3.5 is no longer supported for example, python requires libpython3xxx.so

hiviah, 2021-04-15:
# Several fixes to docs needed

SW versions:

* cocotb version 1.5.1
* Ubuntu 16.04 LTS 64-bit
* Icarus Verilog 11 64-bit
* several Python versions tried, 3.5.2 (system), 3.7.4, 3.9.4 (manually built, in virtualenv)

So multiple issues:

## Python 3.5 no longer supported because of 3.6 syntax

`pip3 install cocotb` no longer works with python 3.5, since it fails on python 3.6 syntax:

    % pip3 install --user cocotb
    Collecting cocotb
      Using cached https://files.pythonhosted.org/packages/e0/be/113b2722dfdf2a44aad58e2ec416ab0a0ae0346bddb0d2cf2e596fcb2a60/cocotb-1.5.1.tar.gz
        Complete output from command python setup.py egg_info:
        Modelsim/Questa executable (vdbg) executable not found. No FLI interface will be available.
        Traceback (most recent call last):
          File "<string>", line 1, in <module>
          File "/tmp/pip-build-k2mm149g/cocotb/setup.py", line 138, in <module>
            "bus": ["cocotb_bus"]
          File "/home/ondro/.local/lib/python3.5/site-packages/setuptools/__init__.py", line 140, in setup
            return distutils.core.setup(**attrs)
          File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
            _setup_distribution = dist = klass(attrs)
          File "/home/ondro/.local/lib/python3.5/site-packages/setuptools/dist.py", line 370, in __init__
            k: v for k, v in attrs.items()
          File "/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
            self.finalize_options()
          File "/home/ondro/.local/lib/python3.5/site-packages/setuptools/dist.py", line 529, in finalize_options
            ep.load()(self, ep.name, value)
          File "/home/ondro/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2325, in load
            return self.resolve()
          File "/home/ondro/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2331, in resolve
            module = __import__(self.module_name, fromlist=['__name__'], level=0)
          File "/tmp/pip-build-k2mm149g/cocotb/.eggs/setuptools_scm-6.0.1-py3.5.egg/setuptools_scm/__init__.py", line 15, in <module>
            from .version import format_version, meta
          File "/tmp/pip-build-k2mm149g/cocotb/.eggs/setuptools_scm-6.0.1-py3.5.egg/setuptools_scm/version.py", line 42
            trace(f"tag '{tag}' parsed to {result}")
                                                  ^
        SyntaxError: invalid syntax

## Python needs to be built so that `libpython3.x.so` is built

Mostly distros are built with it, but with custom Python versions it gets more complicated.

I guess this is enabled with `--enable-shared` in `configure`, complete configure I used was 

    ./configure --prefix=$HOME/local/Python-3.9.4 --enable-shared --enable-loadable-sqlite-extensions --enable-optimizations

Otherwise running fails with weird undecipherable errors like (this is the `simple_dff` example):

    MODULE=test_dff TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog \
            /home/ondro/work/satoshilabs/TASSIC/tools/iverilog/bin/vvp -M /home/ondro/work/satoshilabs/venvs/trezor-python3.7-venv/lib/python3.7/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
         -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib None: None: cannot open shared object file: No such file or directory
         -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
    ERROR: results.xml was not written by the simulation!
    /home/ondro/work/satoshilabs/venvs/trezor-python3.7-venv/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:69: recipe for target 'results.xml' failed
    make[1]: *** [results.xml] Error 1
    make[1]: Leaving directory '/home/ondro/work/satoshilabs/TASSIC/repos/cocotb/examples/simple_dff'
    /home/ondro/work/satoshilabs/venvs/trezor-python3.7-venv/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
    make: *** [sim] Error 2

The `Unable to open lib None` is not exactly helpful, after more digging it needs `libpython3.7m.so`.

As a side note, Python installed with `enable-shared` into non-default location (not /usr/local) also requires you to use `LD_LIBRARY_PATH` so that the libpython.so is found, or use `patchelf` to put rpath in in the python binaries.

ktbarrett, 2021-04-15:
The first error seems to be a bug in setuptools_scm. Report that issue to them.

It does seem like the README is out of date, I'll fix that... but the official documentation is correct. Perhaps "Python development packages" is a bit vague and we can mention the need for libpython.

We use find_libpython to locate the libpython and then directly load it at runtime. The "None" you see is because find_libpython could not find libpython (because you didn't have it installed per your comment). We could add a check to the makefiles to ensure that `LIBPYTHON_LOC` is not set if the result is `None` so it falls back to lookup by library name at runtime.

ktbarrett, 2021-04-16:
Actionable items addressed. Reopen if you think I missed something.

closed 2021-04-16 by ktbarrett