cocotb/cocotb #2527: Ability to run standalone Cocotb tests without the simulator

jshaker000, 2021-04-17:
I've been enjoying cocotb for a few months now. A feature that I think could be helpful would be the ability to perhaps have a python hook to run my tests without invoking the simulator at all.  
   
Would it be possible to use coroutines to have some framework to model HDL, or even just make a dummy DUT so I can check that I'm driving nets the way that I expect and I have no syntax issues and my test "generally works" without needing to elaborate and get a simulation license.  
   
I could imagine this feature essentially being two parts - having some function that could include a list of cocotb tests as a module or something and invoke them - annd then some way to make a 'dummy dut' object that a user could overload and pass to these tests.

Does that seem interesting or helpful - I'd be willing to expand on this idea if anyone finds this interesting. It is just a bit painful to find a python runtime error or an logical error in stimulus after spending a decent amount of time elaborating and loading the VPI and all ..

jshaker000, 2021-04-17:
I guess part of the feasibility of this depends on how much a 'dut' object remembers or learns about itself or if it is essentially just a handle to talk to the simulator ..

eric-wieser, 2021-04-17:
The danger here is that we end up reimplementing a simulator in python just to enable running without a simulator.

> without needing to elaborate and get a simulation license.

You could always use something like `iverilog` side-by-side with your "real" simulator, for testing cocotb without any interesting hardware.

> It also depends on how the event model is handled in cocotb and if python 'knows' anything about the clock - Ie are these coroutines kind of blocking in python because they know the clock is going up and down or are they essentially looping through the simulator to stop and start.

A coroutine that does `await RisingEdge(clock)` is waiting for the simulator to say "hey, a rising edge just happened on clock. Cocotb then finds and resumes the coroutine that was waiting on that. Once the coroutine pauses again, cocotb stops and tells the simulator "ok I'm done responding, resume the simulation".

jshaker000, 2021-04-17:
Right - definately implementing a sexond simulator in python is beyond the scope of cocotb and is definately not worth it.   
  
Perhaps the solution then is to in verilog simply stub out the guts of a module to see if the python is rougly right before elaborating the real thing. hmm

ktbarrett, 2021-04-17:
Do you mind if I convert this into a discussion?

ktbarrett, 2021-04-17:
> Would it be possible to use coroutines to have some framework to model HDL, or even just make a dummy DUT so I can check that I'm driving nets the way that I expect and I have no syntax issues and my test "generally works" without needing to elaborate and get a simulation license.

They call this mocking. I can imagine a use case where you just mock the interface when developing an interface, without having to have a working HDL design to test against or loading a simulator. Since the dut object is just a nested object with basic functionality (get/set/iterate/index), this is something you can make yourself, and seeing how it is specific to a particular design/interface I don't see a way around it. Automating generating of these objects could be done, but I'm not sure if this needs to be in the main repo.

The real problem for us starts with the scheduler and triggers. The triggers really should be refactored to dispatch to the dut objects for implementation. All simulation "bits" can and should be encapsulated in the handles (besides the concept of time/timesteps). The scheduler should already keep track of time step state to be able to check invalid operations (like `ReadWrite` after `ReadOnly`). Adding the concept of time on top of that shouldn't be difficult. For other scheduler bits (like `fork`), I think just refactoring the one cocotb has to be able to run outside of a simulator is feasible. @cmarqu has shown interest in this as well: the ability to use cocotb as a simulation library (in the classical meaning of "simulation" e.g. Simula-67 and various libraries in the same vain). 

> worth trying to support if its not too painful

All possible solutions will be very painful. That doesn't mean it isn't worth doing.

ktbarrett, 2021-04-17:
Also, there is mypy for static checking which may prevent some simpler issues. I have not had any luck with it, but maybe you would. I think adding enough typing information to cocotb interfaces for a user to use mypy if they want is a good goal.

jshaker000, 2021-04-17:
Making this a discussion is fine with me.

That explanation makes compete sense - users can write an arbitrary class that has the methods the dut has and most code would work (perhaps a cocotb geberated dut could have a method "real hardware" that against simulator returns true and otherwise returns false).

But nonetheless there is the issue that scheduling is linked to the simulator so nothing would happen.

I think you outlined well why exactly Id want to use this

ktbarrett, 2021-04-17:
There would have to be a runtime involved when mocking, but it could be made to be simple. The "test" would drive the mocked dut object which would cause edges to occur, the runtime would encode the concept of time, the dut would have `__rising_edge__` special methods that would be used for edges and would set the event that the trigger relied on, and so on.

It would be a simulator, but perhaps a simple one. The issue is that the cocotb codebase is in no shape to accept that currently. Partitioning of responsibility is very poor. The assumption of using an actual HDL simulator over the VPI/VHPI/FLI and *only* that is everywhere. It needs to kick up the level of abstraction a bit.

closed 2021-04-17 by ktbarrett