cocotb/cocotb #2530: When cocotb has more than one test fail It should exit with a non-zero exit code

jshaker000, 2021-04-19:
I use bash scripting to run regressions and it would be helpful if whe nrunning cocotb it returned a non zero exit code if any test failed - rather than needing to parse standard out of the XML.

eric-wieser, 2021-04-19:
It's not possible for cocotb (the python part) to do this - we don't have any way to reliably set the exit code of the simulator. It _would_ be possible to build this into our makefiles, which would go here:

https://github.com/cocotb/cocotb/blob/56d4f9cc4669a318f140d11595d4115bad7ca793/cocotb/share/makefiles/Makefile.inc#L82-L85

Of course, this would amount to having to parse the XML!

Are you using the makefiles, or invoking the simulator with appropriate arguments directly?

jshaker000, 2021-04-19:
I'm using custom flow.

I'm wondering then if cocotb could decide to send $finish() or $fatal() depending on if any tests fail.

If not, I could parse the XML

ktbarrett, 2021-04-19:
We use `vpi_control(vpiFinish, exit_code)`, but it doesn't always respect the exit code =/ Parsing the results.xml is the dependable way to go. Look at `/bin/combine_results.py` for inspiration.

jshaker000, 2021-04-19:
So I might just write a wrapper script to invoke the simulator - capture its exit code, then parse the XML and then somehow combine them to make some exit code.

ktbarrett, 2021-04-19:
I wouldn't mind setting up a `cocotb-config` option to set an exit code given the *expected* path to a results.xml. If it's missing or contains failures, it will print failing tests or missing file error and return non-0. Basically, move that functionality out of the makefile.

jshaker000, 2021-04-19:
I like that idea - I think this is somewhat common functionality needed - people who use the provided makefiles can have it invoke that function and people with custom flow can still invoke that function in their own flow.

imphil, 2021-04-19:
... but please use something other than cocotb-config for this functionality. cocotb-config is like pkg-config or other *-config tools meant to provide information about the cocotb library to other tools at build time.

ktbarrett, 2021-04-19:
> meant to provide information about the cocotb library to other tools at build time.

What's "build time" in this context? We do already have a bunch of makefile helpers in there, is that wrong?

jshaker000, 2021-04-19:
I ended up writing a small ruby check and a hook in our flow after cocotb runs to merge the exit code of the simulator and of this script. I know the whole library is python but maybe this is helpful. This solved my personal issues anyway.
```
#!/usr/bin/env ruby
# Cocotb will (seemingly) never exit with a non-0 exit code, even when a test fails.
# Instead it will output an xml with failures
# Lets parse the xml and then output an appropriate error code ...

require 'nokogiri'
require 'fileutils'
require 'optparse'

options = {
  results_xml: 'results.xml'
}

OptionParser.new do |opts|
  opts.banner = 'Usage parse_cocotb_xml [options]'

  opts.on('-x', '--xml=XML', "Name of the results xml file to parse. Default is #{options.fetch(:results_xml)}") do |x|
    options[:results_xml] = x
  end
end.parse!

xml = Nokogiri::XML(File.open(options[:results_xml], 'r'))
errors = 0
xml.xpath('//testcase').each do |t|
  errors += t.xpath('//failure').size
end

exit(errors == 0 ? 0 : 1)
```

jshaker000, 2021-05-21:
One other thing to highlight is that a python syntax error with the above solution will cause the simulator to exit with a 0 exit code, and then it will leave  the old results.xml and can erroneously let you see a testr as passing. so you actually probably also need a pre flow hook to delete the results.xml

themperek, 2021-05-21:
Sorry for the spam but most of those issues are handled here https://github.com/themperek/cocotb-test