cocotb/cocotb #2533: Error with cocotb 1.5+: Invalid property vpiDefFile for object of type vpiGenScope passed to vpi_get_str()

gretzteam, 2021-04-28:
this was discussed on the gitter. Copying relevant discussion here as the problem seems to be known. This issues acts as a reminder.

cocotb 1.5+ (no issue with cocotb 1.4)
Xcelium 64 bit
Python 3.8 self installed

With cocotb 1.5+ I'm getting this 'error' after the test finishes, right before the final printout . Everything works fine and the test works. 

```
INFO cocotb:          Invalid property vpiDefFile for object of type vpiGenScope passed to vpi_get_str().
INFO cocotb:         /projects/hdl/mydesign.v, 328:
INFO cocotb: 328081.00ns INFO     Test Passed: tone
INFO cocotb: 328081.00ns INFO     Passed 1 tests (0 skipped)
INFO cocotb: 328081.00ns INFO     *************************************************************************
INFO cocotb:                      ** TEST             PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
INFO cocotb:                      *************************************************************************
INFO cocotb:                      ** pdm_cocotb.tone    PASS       328081.00          9.59     34205.27  **
INFO cocotb:                      *************************************************************************
```

Gitter discussion:

```
Kaleb Barrett @ktbarrett Feb 25 20:44
It maybe had something to do with cocotb/cocotb@4d4fffa?
No matter what caused it, when creating a new handle the VPI implementation always queries vpiDefFile on the object rather than doing so lazily. Xcelium apparently hasn't implemented that property on vpiGenScope objects, but the error message it gives is wrong, it's certainly not an invalid property for that type of object.
@marlonjames Any ideas?
A fix for the error message is to make the computation of the filename lazy, but I don't know if it isn't affecting the outcome of your sim if it's worth holding off the 1.5 release.
@gretzteam Can you make an issue about it so we remember to work the issue?

Kaleb Barrett @ktbarrett Feb 25 21:02
Yeah, it's not going to get fixed without a patch to the GPI implementation.
Or to Xcelium...

Kaleb Barrett @ktbarrett Feb 25 21:09
Yeah. It was an improvement in how we discover pseudo-regions with the VPI. The error message was an unintended consequence.
```

marlonjames, 2021-04-28:
Marlon James @marlonjames Feb 25 13:31

> Ok, most objects have `vpiFile` and `vpiLineNo`, which may be what we should be using. Instances also have `vpiDefFile` and `vpiDefLineNo` properties that point to where the definition of the package/interface/program/module is defined.
> Regardless, `vpiGenScope` and `vpiGenScopeArray` don't have any of them because they are simulation constructs.

See also discussion in #1958.

ktbarrett, 2021-04-28:
Would `vpiGenScope` and `vpiGenScopeArray` not have a `vpiFile` and `vpiLineNo`? Generate statements are code construct with a file and line number (I assume that is the start file and lineno?). It should just be saved lexer info.

marlonjames, 2021-04-28:
@ktbarrett You are right. From SV 2017 LRM 37.3.3

> The properties `vpiLineNo` and `vpiFile` can be affected by the \`line compiler directive. See 22.12 for more
details on the \`line compiler directive. These properties are applicable to every object that corresponds to
some object within the source code. The exceptions are objects of the following types:
— vpiCallback
— vpiDelayTerm
— vpiDelayDevice
— vpiInterModPath
— vpiIterator
— vpiTimeQueue
— vpiGenScopeArray
— vpiGenScope

ktbarrett, 2021-04-28:
That's just odd to me. I can understand the other elements on the list, but not `vpiGenScope` or `vpiGenScopeArray`. Oh well 🤷.